orbbec/OrbbecSDK #10: Release Firmware in parallel with SDK releases / where is 1.7.4 compatible firmware?

ofTheo, 2023-09-20:
Currently the only Linux SDK download link on the main Orbbec site is for 1.7.4. 

The Femto Mega 1.1.5 firmware does not work with this SDK version ( neither the Orbbec Viewer or the NetDevice.cpp example work with a 1.1.5 firmware camera for network connection ). 

I had to clone this repo ( @ 1.6.x ) and use the prebuilt 1.6 Orbbec Viewer to get the NetDevice / Network connection working. 
See the discussion here: https://3dclub.orbbec3d.com/t/femto-mega-network/3772/8 

In general it would be super helpful if you could archive SDK versions and firmware versions either here or the main site. 
As once 1.7.4 was release the 1.6 links disappeared from the main site. 

Also do we have an ETA on firmware for Femto Mega that is compatible with 1.7.4 SDK? 

Thanks so much!

hzcyf, 2023-09-21:
1. 由于OrbbecSDK是一个独立运行的项目，其开发和

hzcyf, 2023-09-21:
1. OrbbecSDK is an independent software project, and its development and release plan will not be coupled with specific hardware projects (such as Femto Mega). Hardware projects will have their own plans for firmware update and release.

2. The latest firmware of Femto Mega is version v1.2.6, but it is still in testing and is expected to be released soon. You can see the link on the official website at that time. At present, the officially released version of Femto Mega is only v1.1.5.

3. OrbbecSDK v1.6.3 and v1.7.4 versions, whether downloaded from the official website or obtained in this repo, are compatible with Femto Mega v1.1.5, and both support the function of modifying the device IP. I'm not sure what the specific problem you mentioned is. If it's ok, please introduce the problem you encountered in detail. By the way, uploading the SDK log will be very helpful for us to analyze the problem you encountered.

ofTheo, 2023-09-21:
Thanks @hzcyf - appreciate the explanation for 1) and 2) 

For 3) I built and ran the NetDevice examples both with 1.6 SDK ( this repo ) and 1.7.4 SDK ( from the download SDK page on the orbbec website ).  Here are the logs for both runs. 

Broken 1.7.4: 
[OrbbecSDK-1.7.4-broken.log.txt](https://github.com/orbbec/OrbbecSDK/files/12690037/OrbbecSDK-1.7.4-broken.log.txt)

Working 1.6: 
[OrbbecSDK-1.6-working.log.txt](https://github.com/orbbec/OrbbecSDK/files/12690041/OrbbecSDK-1.6-working.log.txt)

bumbastic, 2023-09-23:
I get the same errors. Looks like the 1.7.4 is not compatible with firmware 1.1.5 and also not compatible with firmware 1.1.7 as stated in the list of compatible devices and firmware versions. I think the problem is the introduction of dhcp, mask and gateway in the OBNetIpConfig. I don't think the older firmware can store these properties. This means that every one buying a Femto Mega will have this problem, they will not be able to use its Ethernet connection. Because the SDK they all are instructed to download is not compatible. This could become someones headache very soon.

hzcyf, 2023-09-25:
@ofTheo @bumbastic Thank you for your feedback. About the problem that the v1.7.4 SDK uses mega v1.1.5 and v1.1.7 firmware broken through network connection, I will push the v1.7.5 SDK fix later.

ofTheo, 2023-09-25:
Thanks for the update @hzcyf! 
Happy to test when it's ready.

hzcyf, 2023-09-27:
@ofTheo @bumbastic  The new version 1.7.5 of OrbbecSDK has been uploaded.  [Release page](https://github.com/orbbec/OrbbecSDK/releases/tag/v1.7.5 )

closed 2023-09-27 by hzcyf

ofTheo, 2023-09-27:
Can confirm that both the 1.7.5 Orbbec Viewer and the 1.7.5 C++ SDK work as expected with the Femto Mega over network connection.