orbbec/OrbbecSDK #9: macOS - uvc_open already opened

ofTheo, 2023-09-20:
Hi,

I just received a Femto Mega and was trying out some of the samples from the SDK. 

For the DepthViewer.cpp the camera is seen but it seems to hang on getVideoStreamProfile. 

```
[2023-09-19 19:33:37.436493][info][37890418][Context.cpp:66] Context created with config: default config!
[2023-09-19 19:33:37.447971][info][37890418][DeviceManager.cpp:562] Found 1 device(s):
[2023-09-19 19:33:37.447983][info][37890418][DeviceManager.cpp:564]   - Name: Femto Mega, PID: 0x0669, SN/ID: CL2K83P001W
[2023-09-19 19:33:37.448009][info][37890418][MacPal.cpp:103] Create PollingDeviceWatcher!
[2023-09-19 19:33:37.448037][info][37890418][Pipeline.cpp:15] Try to create pipeline with default device.
[2023-09-19 19:33:52.262518][warning][37890418][ObUvcDevice.cpp:51] uvc_open  path=2-2-1.2 already opened
[2023-09-19 19:33:52.262867][warning][37890418][ObUvcDevice.cpp:54] uvc_open  path=2-2-1.2 failed,return res:$-3
[2023-09-19 19:33:52.264594][info][37891082][DeviceManager.cpp:117] task finish.
[2023-09-19 19:33:52.264734][info][37890418][Context.cpp:82] Context destroyed
function:Pipeline
args:nullptr
message:uvc_open  path=2-2-1.2 failed,return res-3
type:0
Program ended with exit code: 1
```

This is on a M1 MacBook Pro with MacOs 13.4.1
It seems pretty close but the device has not already been opened. 

Happens no matter what port I use and even if I power cycle the camera. 

Thanks!
Theo

zhonghong322, 2023-09-20:
Femto Mega does not support MacOS，MacOS only support Gemini 2 L、Astra 2、Gemini 2 device.

ofTheo, 2023-09-20:
Thanks! 
Any chance it will be added for macOS? It would make a huge difference to our usage and would open the device up to a lot more users.

ofTheo, 2023-09-20:
Going to close - but would absolutely love if you could get at least network access to Femto Mega working on macOS in the future 😁🙏

henriquenunez, 2023-09-27:
Hey, I am having the same issue with Astra Stereo SU3. What could be the problem?

[image]

henriquenunez, 2023-09-27:
I think this might be due to macOS already locking the UVC device, so the SDK cannot open the stream again.. Any thoughts?

zhonghong322, 2023-09-27:
> Hey, I am having the same issue with Astra Stereo SU3. What could be the problem?
> 
> [image]

This is a permission issue. You can use sudo to open the device with administrator privileges.

henriquenunez, 2023-09-27:
Yes, I actually figured it out some minutes after XDI just got a bit confused, cause the error message says it was already open. Probably it did not distinguish between “couldn’t open because it is locked” and “couldn’t open because I lack privileges”Thx for the reply

henriquenunez, 2023-09-27:
> @henriquenunez - was your solution to run the example with sudo, or did you do something else?

Directly with sudo`sudo ./color_viewer`, for exampleWhat got me confused was that only the UVC (rgb camera was affected) not the depth sensor. Maybe UVC needs to talk to some privileged system API while the other can just send USB packets…

xcy2011sky, 2023-12-23:
@ofTheo   It is expected that by January 30, 2024, FemtoMega will support mac OS and support both USB and network functions

zhonghong322, 2023-12-23:
> @ofTheo It is expected that by January 30, 2024, FemtoMega will support mac OS and support both USB and network functions

We are resolving the issue.

ofTheo, 2023-12-23:
Wow!! That's wonderful news @xcy2011sky @zhonghong322 
Super excited for this - going to be incredible for our community. 🎉🙏

xcy2011sky, 2024-01-17:
@ofTheo 
The latest SDK version 1.9.3 has been released. The camera models FemtoMega, Gemini2, Gemini2L, and Astra2 are all now supported on the MacOS system.

zhonghong322, 2024-01-17:
@ofTheo https://github.com/orbbec/OrbbecSDK/releases/tag/v1.9.3,   The OrbbecViewer tool, which supports Mac, will be released very soon.

ofTheo, 2024-01-19:
@zhonghong322 - awesome to hear about the OrbbecViewer tool. 

Can confirm USB 3.2 works on macOS but does require sudo to launch. 
( This would be amazing to solve ) but I can take a look too. 

[image]

ofTheo, 2024-01-19:
update:  it does seem the USB / sudo situation is fairly flakey on macOS. 
I got it working a couple of times, but then other times it just hangs after listing the device. 

```
DesignIO2023:bin theo$ sudo TrackerDev.app/Contents/MacOS/TrackerDev 
2024-01-19 15:42:26.884 TrackerDev[43718:12163482] WARNING: Secure coding is automatically enabled for restorable state! However, not on all supported macOS versions of this application. Opt-in to secure coding explicitly by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState:.
[warning] ofPixels: setting color not supported yet for UNKOWN format
[01/19 15:42:26.998128][info][12163482][Context.cpp:67] Context created with config: default config!
[01/19 15:42:26.998146][info][12163482][Context.cpp:72] Context work_dir=/users/me/TrackerDev/bin/TrackerDev.app/Contents/Resources
[01/19 15:42:27.000860][info][12163482][MacPal.cpp:104] Create PollingDeviceWatcher!
[01/19 15:42:27.000869][info][12163482][DeviceManager.cpp:15] Current found device(s): (1)
[01/19 15:42:27.000870][info][12163482][DeviceManager.cpp:24] 	- Name: Femto Mega, PID: 0x0669, SN/ID: CL2AA31003T, Connection: USB3.2
```

not sure if this is the macOS sudo issue or something else

zhonghong322, 2024-01-22:
@ofTheo Have you connected the external power adapter to ensure stable power supply? If the power supply is stable, the power indicator on the back of the Femto Mega should be white.

zhonghong322, 2024-02-05:
@ofTheo We have updated to OrbbecSDK version 1.9.4, and the OrbbecViewer tool now supports MacOS.   https://github.com/orbbec/OrbbecSDK/releases/tag/v1.9.4

ofTheo, 2024-02-06:
Thanks @zhonghong322 - I did give the Orbbec Viewer a try on macOS and it requires a lot of steps to even get the application open. 

1 - I first had to expliately allow OrbbecViewer to be run as a non signed app via Security System Preferences. 
2 - Then for each dylib it was trying to load I had to explicitly approve it via the same section in System Preferences. Each time this causes the app to crash so I had to run the app as many times as there was dylib ( 10+ times ). 
3 - Then I had to launch the app with Sudo.

Here you can see the permissions I had to give to each dylib 

[image]

Once I did all above and running with sudo I get a hang after listing the device:

```
DesignIO2023:OrbbecViewer_v1.9.4_202401301118_macos_arm64_release theo$ sudo ./OrbbecViewer 
Password:
2024-02-06 15:01:26.673 INFO  [17124546] [loggerInit@18] **********************************************
2024-02-06 15:01:26.673 INFO  [17124546] [loggerInit@19]  OrbbecViewer launched! Welcome!! 
2024-02-06 15:01:26.673 INFO  [17124546] [loggerInit@20]  	- Version: V1.9.4
2024-02-06 15:01:26.673 INFO  [17124546] [loggerInit@21]  	- Author: 
2024-02-06 15:01:26.673 INFO  [17124546] [loggerInit@22]  	- E-Mail: 
2024-02-06 15:01:26.673 INFO  [17124546] [loggerInit@23]  	- Company: orbbec
2024-02-06 15:01:26.673 INFO  [17124546] [loggerInit@24]  	- Website: http://www.orbbec.com.cn/
2024-02-06 15:01:26.673 INFO  [17124546] [loggerInit@25] **********************************************
2024-02-06 15:01:26.811 INFO  [17124546] [IWindow::IWindow@133] Init glfw down!
[02/06 15:01:26.817745][info][17124546][Context.cpp:69] Context created with config: /Users/theo/Downloads/OrbbecViewer_v1.9.4_202401301118_macos_arm64_release/OrbbecSDKConfig_v1.0.xml
[02/06 15:01:26.817770][info][17124546][Context.cpp:74] Context work_dir=/Users/theo/Downloads/OrbbecViewer_v1.9.4_202401301118_macos_arm64_release
[02/06 15:01:26.817776][info][17124546][Context.cpp:77] 	- SDK version: 1.9.4
[02/06 15:01:26.817781][info][17124546][Context.cpp:78] 	- SDK stage version: main
[02/06 15:01:26.817787][info][17124546][Context.cpp:82] get config EnumerateNetDevice:false
[02/06 15:01:26.817795][info][17124546][MacPal.cpp:36] createObPal: create MacPal!
[02/06 15:01:26.822145][info][17124546][MacPal.cpp:104] Create PollingDeviceWatcher!
[02/06 15:01:26.822164][info][17124546][DeviceManager.cpp:15] Current found device(s): (1)
[02/06 15:01:26.822168][info][17124546][DeviceManager.cpp:24] 	- Name: Femto Mega, PID: 0x0669, SN/ID: CL2AA31003T, Connection: USB3.2
```

If I have the Femto Mega disconnected before launching the app the app opens as expected, but then hangs as soon as I connect the Femto Mega. 

For 1) and 2) I think this could be solved by bundling the dylibs in an App Bundle and getting it Signed and Notarized as a typical Mac App. 

For 3) Does the Mega need the latest firmware? I still have the firmware from when it shipped.

xcy2011sky, 2024-03-14:
@ofTheo
  I hope to communicate with you about the use experience of FemtoBolt. Could you please give me an email address? Or you can contact me directly.My email xuchongyan@orbbec.com;

SimonBiggsUK, 2024-04-01:
I'm following this thread with interest as I would be very keen to have an Orbbec running native to Mac OS. I currently employ an Azure Kinect talking to a Windows GTX 1080 laptop running Kinect SDK in Unity and then send data to an M3 Max Mac over UDP. This works for skeleton data, but I'd love to be able to send video and point cloud data like this - which doesn't seem possible with this setup. On the Mac I am picking up the data in Processing 4.3 (Apple Silicon). Note the latest Mac OS upgrade 14.4.1 seems to have totally broken USB camera input, so not sure if that would represent a further barrier to getting an Orbbec to work with a Mac. I'm having to use Continuity camera to get video input to correlate to the Kinect skeleton data (not a sustainable solution).

xavijulez, 2025-04-05:
> Following up as this is an issue for a lot of other depth camera libraries: See the discussion here: [IntelRealSense/librealsense#11815](https://github.com/IntelRealSense/librealsense/issues/11815)
> 
> The TLDR:
> 
> * Starting in macOS 12 UVC cameras need `sudo` if they are opened via libuvc.
> * This is because using libuvc bypasses Apple's requirement of showing a popup to access a camera ( see: [libuvc on mac can't request camera access libuvc/libuvc#194](https://github.com/libuvc/libuvc/issues/194) )
> * To get round this you can create an empty video class extension with the vid and pid of the product, this prevents Apple locking the device and requiring sudo: https://developer.apple.com/documentation/coremediaio/overriding_the_default_usb_video_class_extension?language=objc
> 
> Reopening, as this will probably affect all supported Orbbec cameras on macOS.

I've tried following the xcode method but turns out i've gotta pay an apple developper plan to create drivers.... Is there any other option? I'm not a developper and I feel completely lost inside Xcode. Is there any simpler way to trick this? 

I'm specifically running the Orbbec Gemini 336L inside Touchdesigner with an M3 Macbook Pro. When I run Touchdesigner on sudo, it works. But it's not an ideal nor long-term solution. Thanks :)