orbbec/OrbbecSDK #12: Orbbec SDK hanging on std::exit

ofTheo, 2023-09-26:
Having a lot of fun with the SDK with the Femto Mega. 
One issue I can't seem to resolve is that when my app closes I get a hang no matter how I seem to close it. 

Is there a rule in terms of what objects need to get destroyed first?
Or what objects you have to hang onto ( as shared_ptr's ) and what ones you can use just for opening the device? 

I should note that my app works great until I try and close it or if I manually call std:exit(0); 
I have tried a lot of different combos, but can't get anything working that isn't a single main(){ style app.  

Below is my app ( simplified ), it also uses GLFW for windowing and setup and update are all on the main thread. 

```
//--------------------------------------------------------------
void ofApp::setup(){

    std::string ip = "192.168.50.79";

    ctx = make_shared<ob::Context>(); 
    auto device = ctx->createNetDevice(ip.c_str(), 8090); //note: also tried holding this as a shared_ptr in ofApp.h 

    // pass in device to create pipeline
    mPipe = std::make_shared<ob::Pipeline>(device);

    // Create Config for configuring Pipeline work
    std::shared_ptr<ob::Config> config = std::make_shared<ob::Config>();

    // Get the depth camera configuration list
    auto depthProfileList = mPipe->getStreamProfileList(OB_SENSOR_DEPTH);
    // use default configuration
    mDepthProfile = depthProfileList->getProfile(0);
    
    // enable depth stream
    config->enableStream(mDepthProfile);
    config->setAlignMode(ALIGN_DISABLE);

    // Pass in the configuration and start the pipeline
    mPipe->start(config);
    
    auto cameraParam = mPipe->getCameraParam();
    pointCloud.setCameraParam(cameraParam);

}

//--------------------------------------------------------------
void ofApp::update(){

    if( mPipe ){
        auto frameSet = mPipe->waitForFrames(100);
        if(frameSet) {
            auto depthFrame = frameSet->getFrame(OB_FRAME_DEPTH);
            if(depthFrame) {
                auto pix = processFrame(depthFrame);
                outputTex.loadData(pix);

                pointCloud.setCreatePointFormat(OB_FORMAT_POINT);
                std::shared_ptr<ob::Frame> frame = pointCloud.process(frameSet); 
                mPointCloudMesh = pointCloudToMesh(frame);
            }
        }
    }

    if( ofGetKeyPressed('c') ){
         if( mPipe ){
            mPipe->stop();
            mPipe.reset();
            std::exit(0); //<------ HANGS. If commented out stream stops without issue. 
        }
    }
}
```

```
#include "ofMain.h"

#include "libobsensor/ObSensor.hpp"
#include "libobsensor/hpp/Error.hpp"
#include <opencv2/opencv.hpp>

class ofApp : public ofBaseApp{

	public:
		void setup();
		void update();
		void draw();
		void exit() override;

		ofPixels processFrame(shared_ptr<ob::Frame> frame);

		void keyPressed(int key);
		void keyReleased(int key);
		void mouseMoved(int x, int y );
		void mouseDragged(int x, int y, int button);
		void mousePressed(int x, int y, int button);
		void mouseReleased(int x, int y, int button);
		void mouseEntered(int x, int y);
		void mouseExited(int x, int y);
		void windowResized(int w, int h);
		void dragEvent(ofDragInfo dragInfo);
		void gotMessage(ofMessage msg);

		std::shared_ptr<ob::Context> ctx;
		//std::shared_ptr<ob::Device> mDevice;
		std::shared_ptr <ob::Pipeline> mPipe;
		std::shared_ptr<ob::StreamProfile> mDepthProfile; 
   		ob::PointCloudFilter pointCloud;

		ofMesh mPointCloudMesh;
		ofTexture outputTex;
		ofEasyCam mCam;
};
```

ofTheo, 2023-09-26:
If I change the closing sequence to have some printouts this is the output from:
```
 if( ofGetKeyPressed('c') ){
         if( mPipe ){
            cout << "1 - STOPPING PIPE" << endl; 
            mPipe->stop();
            mPipe.reset();
            cout << "2 - PIPE STOPPED AND RESET " << endl; 
            ofSleepMillis(100);
            cout << "3 - CALLING exit " << endl; 
            std::exit(0);
        }
    }
```

```
1 - STOPPING PIPE
[2023-09-26 11:41:27.002007][info][63146][Pipeline.cpp:344] Try to stop pipeline!
[2023-09-26 11:41:27.002039][info][63146][Pipeline.cpp:316] Try to stop streams!
[2023-09-26 11:41:27.079484][info][63146][VideoSensor.cpp:657] Video sensor stopped, @OB_SENSOR_DEPTH
[2023-09-26 11:41:27.079494][info][63146][Pipeline.cpp:329] Sensor stream stopped, sensorType=OB_SENSOR_DEPTH
[2023-09-26 11:41:27.079499][info][63146][Pipeline.cpp:335] Stop streams done!
[2023-09-26 11:41:27.080125][info][63146][Pipeline.cpp:373] Stop pipeline done!
[2023-09-26 11:41:27.080134][info][63146][Pipeline.cpp:69] Pipeline destroyed! @0x55A52BD44B60
[2023-09-26 11:41:27.080142][info][63146][FemtoMegaNetDevice.cpp:62] FemtoMega Net device destroyed! PID: 0x0669, SN: CL2K83P001W
[2023-09-26 11:41:27.080155][info][63146][VideoSensor.cpp:265] VideoSensor destroyed, @OB_SENSOR_DEPTH
[2023-09-26 11:41:27.080235][warning][63146][RTSPStreamPort.cpp:189] Stream have not been started!
2 - PIPE STOPPED AND RESET 
3 - CALLING exit 
[2023-09-26 11:41:27.180601][info][63149][DeviceManager.cpp:117] task finish.
```

Note: I have tried not closing the pipe or resetting too.

hzcyf, 2023-09-27:
I believe the issue is related to a deadlock caused by the code I wrote to prevent the spdlog crash problem. The registry instance of spdlog is a static object, and it needs to be ensured that no log output remains after the object is destroyed when the program exits. Therefore, I cached the object and added a lock to avoid this situation. This issue has been resolved in the recent version. May you can verifiy it using v1.7.5.

ofTheo, 2023-09-27:
Ah great. Thank you @hzcyf 
I’ll give this and #10 a try tomorrow and will report back. 🙏

ofTheo, 2023-09-27:
Awesome - just tried 1.7.5 ( #13 ) and it totally fixes this issue. 
Thanks @hzcyf

closed 2023-09-27 by ofTheo