orbbec/OrbbecSDK #14: Point cloud flipped in Y with 1.6.x -> 1.7.5 SDK

ofTheo, 2023-09-27:
Sorry to open another issue and this might just be an intentional change ( as I only see this in the SDK not the Orbbec Viewer ). 

But with identical code that displays the point cloud, switching from 1.6.x to 1.7.5 I see the point cloud inverted in Y. 

Screenshot from app using 1.6.x SDK: 
[image]

Screenshot from the same app using 1.7.5. SDK: 
[image]

Feel free to close if this was an intentional change. 
Just wanted to post it incase it was a regression.

hzcyf, 2023-10-07:
Close as it is  an intentional change.

closed 2023-10-07 by hzcyf

ofTheo, 2023-10-07:
That makes sense! 
I noticed that it’s the same orientation now as the Kinect Azure so that does make it easier to port old code over. 

Thanks!!