orbbec/OrbbecSDK #17: Hardware trigger model for Femto Mega

moversevlad, 2023-10-09:
Hello, I m trying to setup a multi-view system and get the *hardware* synchronized streams.
The idea is to have 1 device set in *OB_MULTI_DEVICE_SYNC_MODE_PRIMARY* and the rest of them in the *OB_MULTI_DEVICE_SYNC_MODE_HARDWARE_TRIGGERING* mode, but I get an error when trying to set up the latter.

How would be the correct way to go about it?

hzcyf, 2023-10-10:
The Mega device only supported those modes: Free Run, Standalone, Primary, Secondary. You can get  the supported multi device sync mode bitmap of your device by call `Device::getSupportedMultiDeviceSyncModeBitmap()` function

closed 2023-10-11 by hzcyf

moversevlad, 2023-10-11:
Thank you for your response. 
When I am trying to get my app running, I get the secondary device in false state (orange blinking) and I dont know what is going on with it. Can you give me some pointers to look into it?
Can you clarify to me if its safe to have 2 instances of ob::Context?
Currently my  app has a class for each device with its own context.
Also, should every device run on its own thread, or not?
Thanks