orbbec/OrbbecSDK #21: Femto Mega timestamps not synchronization when working as POE devices

moversevlad, 2023-10-19:
I've a setup of 2 Femto Mega devices and I use a syncbox to synchronize them. When I am using the USB interface, the hardware timestamps seem to be updated despite the fact that hardware timestamp seem not to match
[image]

but when I am using the POE interface, a) hardware timestamps are not synced to host PC b) system timestamps are also off
[image]

(I am moving a retro-reflective object in the scene at high speed, this is what you see in the pictures)

hzcyf, 2023-10-23:
The v1.1.5 version of Femto Mega device firmware can only obtain the startup time of the device through POE connection. This issue can be resolved by updating the firmware to v1.2.6.

moversevlad, 2023-10-30:
Hello, on top of this question. Lets suppose we're able to synchronize the cameras and the timestamps of the devices are synced with the host PC. 
After that we want to group frames across all cameras, based on the timestamps. Should I use device timestamps or system timestamps?

hzcyf, 2023-11-01:
> Hello, on top of this question. Lets suppose we're able to synchronize the cameras and the timestamps of the devices are synced with the host PC. After that we want to group frames across all cameras, based on the timestamps. Should I use device timestamps or system timestamps?

Using device timestamp is better, which is the timestamp when the data frame is exposed, while system timestamp is the timestamp after the data frame is received by the SDK.

moversevlad, 2023-11-01:
Amazing, thank you.
It seems that I get quite some frames that do not match their device timestamps.
[image]
This is an example of this case. Previous too that frame, I get HW synced frames, both on what I see, as well as on their device timestamps.
Seems like I get quite high percentage of frames that are not synced, is there such an issue in the backlog  /  have anyone observed such behaviour?

zhonghong322, 2023-12-23:
Please download the firmware, upgrade tool, and upgrade documentation, and then follow the upgrade documentation to perform firmware upgrades.
https://github.com/orbbec/OrbbecFirmware/releases/tag/Femto-Mega-Firmware