orbbec/OrbbecSDK #23: Please compile the version number into OrbbecSDK.dll

dbruning, 2023-10-21:
It would simplify development and support if you could compile the DLL version into orbbecSDK.dll. Right now we need to check file dates & sizes to figure out which version we're working with.

hzcyf, 2023-10-23:
Thank you for your suggestion! We will consider it in future versions. Do you have any recommended solutions, such as directly writing the version information in the dll file name, just like the Linux version?

dbruning, 2023-10-30:
I'm not sure how it versioning genrally works for Linux, but on Windows it's possible to compile the version number directly into metadata in the .dll. I'm not a C/C++ developer so I can't tell you exactly but I'm sure it's just a setting in Visual Studio.

hzcyf, 2023-11-01:
> I'm not sure how it versioning genrally works for Linux, but on Windows it's possible to compile the version number directly into metadata in the .dll. I'm not a C/C++ developer so I can't tell you exactly but I'm sure it's just a setting in Visual Studio.

Thanks for your reply！ We have added the version info to metadata of dll file by CMake. It will be enabled in the next version

closed 2023-11-01 by hzcyf