orbbec/OrbbecSDK #24: net enumeration doesn't work with femto mega

derivativerobmc, 2023-10-24:
I'm trying to use the ob_enable_net_device_enumeration command with my Femto Mega camera, but I don't get any response from the camera. I can see the GVCP packets being broadcast, and I can connect to my camera with the IP address, but it won't show up in the device list. This happens with the Orbbec viewer app or my own code.

I don't know if this is because my camera is still 1.1.5 firmware, or if the feature isn't expected to work. I tried to update my firmware using the viewer app, but I can't find the newer version.

hzcyf, 2023-10-25:
Femto Mega needs to be updated to v1.2.6 firmware to support this feature. The version of the firmware will be released in the near future, and I will post the link here when it is available.

tim-depthkit, 2023-11-08:
Any updates on when a new firmware will be released for the Femto Mega?

zhonghong322, 2023-12-23:
Please download the firmware, upgrade tool, and upgrade documentation, and then follow the upgrade documentation to perform firmware upgrades.
https://github.com/orbbec/OrbbecFirmware/releases/tag/Femto-Mega-Firmware         @tim-depthkit @derivativerobmc

closed 2024-05-14 by hzcyf