orbbec/OrbbecSDK #29: Femto Mega 3840 x 2160 resolution on ethernet.

akashjinandra, 2023-11-02:
We are trying to use the Femto mega using the Ethernet interface at 3840 by 2160 resolution for color. When using the USB C interface for the camera I am able to get the full resolution of 3840 by 2160 using the MJPG format. However when I try to do this with the Ethernet Interface it doesn't allow me to use that format and forces me to use the H264 format, however the orbbec sdk doesn't seem to support this. 

Is there any plan to make this work? We would really like to use the ethernet interface vs the USB to save bandwidth.

hzcyf, 2023-11-03:
Femto Mega v1.2.6 firmware supports MJPG format. The firmware of this version will be released soon.

However, due to the limitation of RTSP/RTP protocol, the maximum resolution supported by MJPG format is only 1080p. Read this: [3.1 JPEG header](https://www.rfc-editor.org/rfc/rfc2435#section-3.1).

akashjinandra, 2023-11-03:
That is unfortunate, however the Orbbec viewer allows me to view the 3840 by 2160 in h264 format, will you guys allow for this in your SDK?

hzcyf, 2023-11-06:
The current SDK and firmware version v1.1.5 already support this format on Ethernet mode, and OrbbecViewer is developed based on the output of this format provided by the SDK.

Tips: The SDK is not responsible for decoding H264, you need to complete the decoding yourself, you can use third-party libraries such as ffmpeg to complete this task.  You can refer to this: https://github.com/orbbec/OrbbecSDK/issues/16