orbbec/OrbbecSDK #30: What is "Write buffer overflow!"?

cj-yoon, 2023-11-03:
Hello.
We are using Gemini-E camera for depth detection.
However, it has problem that the image buffer cannot received sometimes.
The OrbbekSDK log in these situation is as follows.

```
[1970-01-01 09:12:30.871355][warning][2910][OpenNIDeviceInfo.cpp:179] New openni device matched.
[1970-01-01 09:12:30.871416][info][2910][DeviceManager.cpp:562] Found 2 device(s):
[1970-01-01 09:12:30.871423][info][2910][DeviceManager.cpp:564]   - Name: Gemini E, PID: 0x065C, SN/ID: [PRIVATE]
[1970-01-01 09:12:30.871477][info][2910][DeviceManager.cpp:564]   - Name: Gemini E, PID: 0x065C, SN/ID: [PRIVATE]
[1970-01-01 09:12:30.871570][info][2910][LinuxPal.cpp:109] Create PollingDeviceWatcher!
[1970-01-01 09:12:30.873247][info][2910][OpenNIHostProtocol.cpp:528] Hardware versions: FW=5.8.23 (14), HW=0, Chip=7, Sensor=0, SYS=12
[1970-01-01 09:12:30.874439][error][2910][OpenNIHostProtocol.cpp:545] Get usb core type failed!
[1970-01-01 09:12:30.882812][info][2910][OpenNISensorFirmware.cpp:1020] Sensor serial number:[PRIVATE]
[1970-01-01 09:12:30.885563][info][2910][OpenNISensorFirmware.cpp:1048] Firmware version RD3480
[1970-01-01 09:12:30.886888][info][2910][OpenNISensorFirmware.cpp:1054] Device frequency 31.25
[1970-01-01 09:12:31.297756][info][2910][OpenNISensorFirmware.cpp:145] OpenNI2 camera support Watchdog function.
[1970-01-01 09:12:31.298236][info][2910][OpenNIDevice.cpp:29] OpenNI device created! PID: 0x065c, SN: [PRIVATE]
[1970-01-01 09:12:31.298258][info][2910][DeviceManager.cpp:481] Device created successfully! Name: Gemini E, PID: 0x065c, SN/ID: [PRIVATE]
[1970-01-01 09:12:31.299143][info][2910][OpenNIDevice.cpp:358] Depth sensor has been created!
[1970-01-01 09:12:31.327106][info][2910][OpenNIDevice.cpp:417] Color sensor has been created!
[1970-01-01 09:12:31.330164][info][2910][OpenNIDevice.cpp:464] IR sensor has been created!
[1970-01-01 09:12:31.334694][info][2910][Pipeline.cpp:42] Pipeline created with device: {name: Gemini E, sn: [PRIVATE]}, @0x55A2AF7A00
[1970-01-01 09:12:31.338662][warning][2910][AbstractDevice.cpp:840] Get supported D2C profile is null!
[1970-01-01 09:12:31.338706][info][2910][Pipeline.cpp:236] Try to start streams!
[1970-01-01 09:12:31.338780][info][2910][VideoSensor.cpp:554] start OB_SENSOR_COLOR stream with profile: {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB, width: 640, height: 480, fps: 15}
[1970-01-01 09:12:31.844951][info][2910][Pipeline.cpp:249] Start streams done!
[1970-01-01 09:12:31.844965][info][2910][Pipeline.cpp:232] Pipeline start done!
[1970-01-01 09:12:31.965057][info][2910][OpenNIHostProtocol.cpp:528] Hardware versions: FW=5.8.23 (14), HW=0, Chip=7, Sensor=0, SYS=12
[1970-01-01 09:12:31.974574][error][2910][OpenNIHostProtocol.cpp:545] Get usb core type failed!
[1970-01-01 09:12:32.009740][info][2910][OpenNISensorFirmware.cpp:1020] Sensor serial number:[PRIVATE]
[1970-01-01 09:12:32.014360][info][2910][OpenNISensorFirmware.cpp:1048] Firmware version RD3480
[1970-01-01 09:12:32.017080][info][2910][OpenNISensorFirmware.cpp:1054] Device frequency 31.25
[1970-01-01 09:12:32.401661][info][2910][OpenNISensorFirmware.cpp:145] OpenNI2 camera support Watchdog function.
[1970-01-01 09:12:32.401797][info][2910][OpenNIDevice.cpp:29] OpenNI device created! PID: 0x065c, SN: [PRIVATE]
[1970-01-01 09:12:32.401807][info][2910][DeviceManager.cpp:481] Device created successfully! Name: Gemini E, PID: 0x065c, SN/ID: [PRIVATE]
[1970-01-01 09:12:32.402078][info][2910][OpenNIDevice.cpp:358] Depth sensor has been created!
[1970-01-01 09:12:32.420133][info][2910][OpenNIDevice.cpp:417] Color sensor has been created!
[1970-01-01 09:12:32.427585][info][2910][OpenNIDevice.cpp:464] IR sensor has been created!
[1970-01-01 09:12:32.454293][info][2910][Pipeline.cpp:42] Pipeline created with device: {name: Gemini E, sn: [PRIVATE]}, @0x55A2FB2E90
[1970-01-01 09:12:32.463331][warning][2910][AbstractDevice.cpp:840] Get supported D2C profile is null!
[1970-01-01 09:12:32.463354][info][2910][Pipeline.cpp:236] Try to start streams!
[1970-01-01 09:12:32.463382][info][2910][VideoSensor.cpp:554] start OB_SENSOR_COLOR stream with profile: {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB, width: 640, height: 480, fps: 15}
[1970-01-01 09:12:32.963579][info][2910][Pipeline.cpp:249] Start streams done!
[1970-01-01 09:12:32.963614][info][2910][Pipeline.cpp:232] Pipeline start done!
[1970-01-01 09:12:36.865647][warning][3452][Pipeline.cpp:289] Pipeline source frameset queue fulled, drop the oldest frame!
[1970-01-01 09:12:37.885774][error][3801][OpenNIPacketProcessor.cpp:18] Write buffer overflow!
[1970-01-01 09:12:37.906508][warning][3791][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor15FormatConverterE@0x55a2fa6090
[1970-01-01 09:12:37.962194][warning][3419][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor15FormatConverterE@0x55a2aeaa70
[1970-01-01 09:12:38.789634][error][3426][OpenNIPacketProcessor.cpp:18] Write buffer overflow!
[1970-01-01 09:12:39.159428][error][3801][OpenNIPacketProcessor.cpp:18] Write buffer overflow!
[1970-01-01 09:12:39.917928][warning][3526][Pipeline.cpp:289] Pipeline source frameset queue fulled, drop the oldest frame! [**16 logs in 3052ms**]
[1970-01-01 09:12:40.424876][error][3801][OpenNIPacketProcessor.cpp:18] Write buffer overflow!
[1970-01-01 09:12:40.925246][warning][4311][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor15FormatConverterE@0x55a2fa6090 [**8 logs in 3018ms, last: 09:12:38.205768**]
[1970-01-01 09:12:40.962774][warning][3875][Pipeline.cpp:289] Pipeline source frameset queue fulled, drop the oldest frame!
[1970-01-01 09:12:41.026993][warning][4314][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor15FormatConverterE@0x55a2aeaa70 [**21 logs in 3064ms, last: 09:12:40.853725**]
[1970-01-01 09:12:41.855024][error][3801][OpenNIPacketProcessor.cpp:18] Write buffer overflow!
[1970-01-01 09:12:41.857410][error][3426][OpenNIPacketProcessor.cpp:18] Write buffer overflow!
[1970-01-01 09:12:42.005588][error][3801][OpenNIPacketProcessor.cpp:18] Write buffer overflow!
[1970-01-01 09:12:43.587833][warning][3854][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor15FrameSoftFilterE@0x55a2d88c90
[1970-01-01 09:12:44.087362][warning][4485][Pipeline.cpp:289] Pipeline source frameset queue fulled, drop the oldest frame! [**11 logs in 3124ms, last: 09:12:43.342218**]
[1970-01-01 09:12:44.417056][warning][3451][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor30Disparity2DepthConverterHalleyE@0x55a2ae4af0
[1970-01-01 09:12:45.932297][warning][3526][Pipeline.cpp:289] Pipeline source frameset queue fulled, drop the oldest frame! [**52 logs in 6014ms**]
[1970-01-01 09:12:46.948059][warning][4656][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor15FrameSoftFilterE@0x55a2d88c90 [**19 logs in 3360ms, last: 09:12:46.392565**]
[1970-01-01 09:12:46.975627][warning][3791][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor15FormatConverterE@0x55a2fa6090 [**30 logs in 6050ms**]
[1970-01-01 09:12:47.052407][warning][3419][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor15FormatConverterE@0x55a2aeaa70
[1970-01-01 09:12:47.438268][error][3801][OpenNIPacketProcessor.cpp:18] Write buffer overflow!
[1970-01-01 09:12:47.522890][warning][4829][FrameProcessingBlock.cpp:89] Source frameset queue fulled, drop the oldest frame! N11libobsensor30Disparity2DepthConverterHalleyE@0x55a2ae4af0 [**1 logs in 3105ms, last: 09:12:44.520587**]
[1970-01-01 09:12:48.223398][error][3801][OpenNIPacketProcessor.cpp:18] Write buffer overflow!
```

We want to know the meaning of "Write buffer overflow!" at these log. And, if possible, We would like to see the source code of OpenNIPacketProcessor.cpp.

Thanks a lot.

hzcyf, 2023-11-06:
The write buffer overflow is caused by the loss of critical information during frame assembly, resulting in the combination of data packets from multiple frames into a single one. This issue may be exacerbated by an underperforming CPU or other programs occupying system resources, preventing the SDK from processing data in a timely manner as indicated by other log entries.

cj-yoon, 2023-11-06:
Thank you.

You said that was the loss of information during frame assembly from multiple frames, right?
I have another question about that. If we use only one frame, then this problem is fixed?

Currently, we used two types of frames; color and depth. If I used only depth frame, then this problem will be fixed?

hzcyf, 2023-11-07:
During USB transmission of frame data, it is divided into multiple packets. If the reception of packets is delayed, some packets may be lost. If a lost packet contains critical information, such as the frame end marker, it may cause the SDK to incorrectly treat the next frame's packet as part of the current frame when reassembling the data frame from received packets, resulting in a buffer overflow error. 

Therefore, even if you only used depth, this situation can still occur if the host's performance is insufficient. However, since only depth is used, the process of receiving and processing color data will be skipped, allowing more performance  to depth. In theory, this should alleviate the problem. It is recommended to conduct actual tests for confirmation.

cj-yoon, 2023-11-10:
OK, Thank you. I'll try it. If the error occurs after adjusting it, I'll reopen this issue.

closed 2023-11-10 by cj-yoon