orbbec/OrbbecSDK #34: How to get imu->camera extrinsics?

kaatrasa, 2023-12-05:
Hey, I have Orbbec Femto Mega and Astra 2. Is there a way to get imu->camera extrinsics from the OrbbecSDK API? I noticed that you have [OrbbecSDK-K4A-Wrapper](https://github.com/orbbec/OrbbecSDK-K4A-Wrapper) that has this option, but we would like to use the OrbbecSDK.

zhonghong322, 2023-12-11:
Thank you for the feedback, we will fix this issue in future versions.

zhonghong322, 2024-01-17:
@kaatrasa https://github.com/orbbec/OrbbecSDK/releases/tag/v1.9.3

kaatrasa, 2024-01-17:
Thanks! Do you plan to add support for other devices in future releases?

zhonghong322, 2024-01-17:
Yes, we will address any reasonable user requirements or issues.

kaatrasa, 2024-01-17:
Sorry, I meant more specifically, will other devices get imu->camera extrinsics support in future? (other than Femto Mega)

zhonghong322, 2024-01-17:
Does the extrinsic between imu to depth for Mega meet your requirements? Regarding the extrinsics for Astra2, I will provide feedback to our product team.

kaatrasa, 2024-01-17:
> Does the extrinsic between imu to depth for Mega meet your requirements?

Yes, the `imu->camera` calibration from OrbbecSDK API seems very similar to what we get from our own calibration scripts for Femto Mega :+1: 

>  Regarding the extrinsics for Astra2, I will provide feedback to our product team.

Thanks again! Having this support for all device models would be great, because otherwise our users need to do the `imu->camera` calibration themselves, if they don't happen to be using Femto Mega or Astra2 (devices we have currently tested).

closed 2024-01-17 by kaatrasa

hplegend, 2024-10-29:
> line->start(config);

Dose python obrbecSDK surpport get IMU-depth extrinsics?