orbbec/OrbbecSDK #37: Femto Mega cameras and IMU have inconsistent device timestamps

kaatrasa, 2023-12-13:
Hey, `ob::ColorFrame::timeStampUs()` and `ob::DepthFrame::timestampUs()` return this type of timestamps `1623497648.304`, whereas `AccelFrame/GyroFrame::timestampUs()` start from `0` when the device is started (i.e. they are using different clock/are inconsistent). In addition, the timestamps seem to have 1 millisecond resolution (i.e. all timestamps seem to be rounded to closest millisecond). I also tested Astra 2 and it did **not** have this issue.

Some additional questions:
1) Does the image device timestamp correspond to start/middle/end of image exposure? We would like to get middle point of exposure
2) Is there way to get `ob::ColorFrame` exposure time? (not so critical if timestamp is middle point of exposure).
3) IMU samples are batched (i.e. ~10 gyro or accel samples arrive at once). Ideally we would disable this behaviour (for instance OAK-D devices have option to set IMU batch size). This also happens on Astra 2.

Here is an example of the IMU & RGB-D data I have serialized in case it is helpful:
```
"sensor":{"type":"gyroscope","values":[0.002128450432792306,-0.015963377431035042,0.003192675532773137]},"time":10.658999999999999}
{"sensor":{"type":"gyroscope","values":[0.006385351065546274,-0.018091827630996704,0.002128450432792306]},"time":10.66}
{"sensor":{"type":"gyroscope","values":[0.002128450432792306,-0.015963377431035042,0.002128450432792306]},"time":10.661}
{"sensor":{"type":"gyroscope","values":[0.004256900865584612,-0.01915605366230011,0.001064225216396153]},"time":10.661999999999999}
{"sensor":{"type":"gyroscope","values":[0.002128450432792306,-0.015963377431035042,0.004256900865584612]},"time":10.663}
{"sensor":{"type":"gyroscope","values":[0.004256900865584612,-0.015963377431035042,0.003192675532773137]},"time":10.664}
{"sensor":{"type":"gyroscope","values":[0.006385351065546274,-0.017027603462338448,0.002128450432792306]},"time":10.665}
{"sensor":{"type":"gyroscope","values":[0.004256900865584612,-0.018091827630996704,0.003192675532773137]},"time":10.666}
{"sensor":{"type":"gyroscope","values":[0.005321125965565443,-0.015963377431035042,0.005321125965565443]},"time":10.667}
{"sensor":{"type":"gyroscope","values":[0.004256900865584612,-0.01915605366230011,0.005321125965565443]},"time":10.668}
{"sensor":{"type":"accelerometer","values":[0.2679687440395355,0.00957031175494194,10.048828125]},"time":10.658999999999999}
{"sensor":{"type":"accelerometer","values":[0.21054686605930328,-0.00478515587747097,10.020116806030273]},"time":10.66}
{"sensor":{"type":"accelerometer","values":[0.24404296278953552,-0.03828124701976776,9.98183536529541]},"time":10.661}
{"sensor":{"type":"accelerometer","values":[0.24404296278953552,-0.00478515587747097,10.0009765625]},"time":10.661999999999999}
{"sensor":{"type":"accelerometer","values":[0.22490233182907104,0.0,10.015331268310547]},"time":10.663}
{"sensor":{"type":"accelerometer","values":[0.22490233182907104,0.00957031175494194,10.0009765625]},"time":10.664}
{"sensor":{"type":"accelerometer","values":[0.24882811307907104,0.014355468563735485,10.010546684265137]},"time":10.665}
{"sensor":{"type":"accelerometer","values":[0.24404296278953552,0.00478515587747097,9.98183536529541]},"time":10.666}
{"sensor":{"type":"accelerometer","values":[0.24882811307907104,0.0,9.996191024780273]},"time":10.667}
{"sensor":{"type":"accelerometer","values":[0.2679687440395355,0.0,10.00576114654541]},"time":10.668}
{"frames":[{"cameraInd":0,"colorFormat":"rgb","height":720,"type":"depth","width":1280},{"aligned":false,"cameraInd":1,"colorFormat":"gray16","depthScale":0.001,"features":[],"hasDepth":true,"height":576,"number":1,"type":"depth","width":640}],"number":1,"time":1623497648.304}
```

zhonghong322, 2023-12-18:
1、Femto Mega cameras and IMU have inconsistent device timestamps
We are using firmware version 1.2.7 and SDK version 1.8.3, and the tests are successful. If the time synchronization interface is called, the device timestamp will be in the format of 1702903353444 (ms). If the interface is not called, the device timestamp starts counting from 0. However, regardless of the scenario, the timestamps remain consistent.

[image]

2、Does the image device timestamp correspond to start/middle/end of image exposure? We would like to get middle point of exposure
start image exposure。

3、Is there way to get ob::ColorFrame exposure time? (not so critical if timestamp is middle point of exposure).
not support 

4、IMU samples are batched (i.e. ~10 gyro or accel samples arrive at once). Ideally we would disable this behaviour (for instance OAK-D devices have option to set IMU batch size). This also happens on Astra 2.

The batch size is related to the sampling frequency of the IMU and cannot be adjusted.

kaatrasa, 2023-12-18:
Hey, thanks for the reply!

I am also using SDK 1.8.3, but the firmware version is 1.1.5.
[image]

How can I upgrade the firmware, OrbbecViewer shows this:
[image]

zhonghong322, 2023-12-20:
> Hey, thanks for the reply!
>
> I am also using SDK 1.8.3, but the firmware version is 1.1.5. [...]
Please download the firmware, upgrade tool, and upgrade documentation, and then follow the upgrade documentation to perform firmware upgrades.
https://github.com/orbbec/OrbbecFirmware/releases/tag/Femto-Mega-Firmware

zhonghong322, 2023-12-20:
Please download the firmware, upgrade tool, and upgrade documentation, and then follow the upgrade documentation to perform firmware upgrades.
https://github.com/orbbec/OrbbecFirmware/releases/tag/Femto-Mega-Firmware

kaatrasa, 2023-12-21:
Thanks, that fixed the imu/camera timestamp sync issue! 

FYI: I noticed that the timestamps still have 1millisecond accuracy/resolution even if I use the 2kHz imu sample rate:
[image]

zhonghong322, 2023-12-22:
yes，imu has only 1millisecond accuracy right now.

closed 2023-12-22 by kaatrasa