orbbec/OrbbecSDK #39: Undistorted point cloud from wide FoV

zacharylr, 2023-12-19:
I’ve been working on a software that perform blob detection using the point clouds from Kinect and I’m working on updating it to work with the latest Orbbec cameras (Femto Bolt and Femto Mega)

I started to get the point cloud from the wide field of view (512 x 512) and when I do, the point cloud is distorted and not properly aligned with real world coordinates.
(A in the image)

To remedy that, Orbbec has a mode to align the depth points with the color camera view (D2C). However, the points that are not within the camera view are discarded, rendering the use of the wide FoV texture useless because a big part of the data is lost.
(B in the image)

I was wondering if there's a way to undistort the point cloud based on world coordinates, like the Azure Kinect SDK had (C in the image), and not based on the RGB camera view? Should I just use the K4A wrapper instead of the Orbbec SDK?

[image]

Thank you!

zhonghong322, 2023-12-20:
1. I started to get the point cloud from the wide field of view (512 x 512) and when I do, the point cloud is distorted and not properly aligned with real world coordinates.

yes ，the images are not undistorted。

2. To remedy that, Orbbec has a mode to align the depth points with the color camera view (D2C). However, the points that are not within the camera view are discarded, rendering the use of the wide FoV texture useless because a big part of the data is lost.

yes，Due to the difference in FOV (Field of View) between the Depth camera and the Color camera, after applying the D2C (Depth-to-Color) alignment, some Depth data will be cropped or discarded.

zacharylr, 2023-12-20:
Thank you for the quick answer.

Does this mean that there is no way to access an undistorted point cloud that has all the data from the wide FoV depth?
Is there a plan to integrate this in the OrbbecSDK or should I just use the K4A wrapper?

Thank you!

zhonghong322, 2023-12-21:
> Thank you for the quick answer.
> 
> Does this mean that there is no way to access an undistorted point cloud that has all the data from the wide FoV depth? Is there a plan to integrate this in the OrbbecSDK or should I just use the K4A wrapper?
> 
> Thank you!

There is no plan to do distortion correction inside the Orbbec SDK for the Femto Mega and Femto Bolt.   K4a Wrapper implementation is similar to azure kinect, and D2C, C2D are all using azure kinect's code. You can test it to see if it meets your needs.

zacharylr, 2023-12-21:
This is a feature we've been using for a while with the AzureKinectSDK and it would be really useful if it would be implemented in the OrbbecSDK since we are updating a lot of our projects with Orbbec devices. The function _k4a_calibration_2d_to_3d()_ from the AzureKinectSDK is what is used to perform this.
[image]

 Could you re-evaluate the possibility of integrating this in the OrbbecSDK so we could have access to the full undistorted wide FoV depth point cloud? We would like to avoid using the K4Awrapper as we intend on using Femto Mega cameras connected via Ethernet.
Thank you

ofTheo, 2023-12-22:
Just want to chime in that a LOT of people in our [openFrameworks](http://github.com/openframeworks/openFrameworks) community are using the point cloud of the Kinect Azure to do body / real world tracking.   

A mode that accurately returns the point cloud data in world coordinates ( XYZ mm ) without distortion is really essential for us. 

We are in the middle of porting a bunch of projects over that will use 7+ Femto Mega devices per install and we really need feature parity of the Kinect, especially when returning world coordinates. 

Can we please get a `ALIGN_D2W_MODE`  ( depth to world mode )  🙏

zhonghong322, 2023-12-23:
Femto Mega and Femto Bolt Undistorted，We will consider it in future versions of the Orbbec SDK.

zacharylr, 2024-01-08:
This is a good thing to read as this feature is primordial for us.
Woud you have an ETA on when this feature could be implemented?
Thank you

zhonghong322, 2024-01-09:
The SDK version is expected to be released sometime this month.

zhonghong322, 2024-01-09:
> This is a feature we've been using for a while with the AzureKinectSDK and it would be really useful if it would be implemented in the OrbbecSDK since we are updating a lot of our projects with [...]

If the firmware is upgraded to version 1.2.7 or above and using the K4A Wrapper, Femto Mega also supports connecting devices through network connections. 

You can copy the OrbbecSDKConfig_v1.0.xml file from the OrbbecSDK repository（https://github.com/orbbec/OrbbecSDK/tree/1.8.x/misc/config/OrbbecSDKConfig_v1.0.xml） and paste it into the same directory as the OrbbecSDK.dll (for Windows) or libOrbbecSDK.so (for Linux).  Then, refer to the provided screenshot to modify the configuration file."

[image]

zhonghong322, 2024-01-09:
If using the network functionality of the Femto Mega with the K4a Wrapper, please note the following:

zacharylr, 2024-01-09:
> The Orbbec SDK version is expected to be released sometime this month.

Just to make sure I understand, the feature we've been mentionning (point cloud data in world coordinates ( XYZ mm ) without distortion ) will be available in the next release this month?

zhonghong322, 2024-01-17:
@zacharylr OrbbecSDK 1.9.3 (https://github.com/orbbec/OrbbecSDK/releases/tag/v1.9.3 ) added a general utility class (CoordinateTransformHelper), which supports point transformation between different coordinate systems, D2C conversion, and distortion correction for depth point clouds and RGBD point clouds.

* Here is an example of how to invoke the functions of the CoordinateTransformHelper class (examples\cpp\Sample-Transformation)

ofTheo, 2024-01-17:
Wonderful - thank you @zhonghong322  

Just to check, to get undistorted point clouds, should we be using: 
`transformationDepthToPointCloud`  

https://github.com/orbbec/OrbbecSDK/blob/10823024d7947a24846acf32ac4092e836f2d727/include/libobsensor/hpp/Utils.hpp#L125C17-L125C48

zhonghong322, 2024-01-18:
@ofTheo yes, Here is an example of how to invoke the functions of the CoordinateTransformHelper class (examples\cpp\Sample-Transformation)

hzcyf, 2024-02-23:
Closed as resolved.

closed 2024-02-23 by hzcyf

mightypanda, 2024-10-10:
Hello @ofTheo @zhonghong322 ,
We need to use this function in the python pyorbbecsdk, is it included or can it be included modifying the sdk source?

Thank you