orbbec/OrbbecSDK #40: Femto Mega duplicate IMU timestamps

kaatrasa, 2023-12-22:
I noticed a bug where Femto Mega returns duplicate timestamps for accel & gyro even if I use 200Hz mode. Ubuntu 20.04, SDK 1.8.3, firmware 1.2.7.

I also tried on Astra2, but did not get duplicate timestamps even in 1KHz mode. Also, it would be great if you could fix the issue where IMU timestamps have 1ms accuracy; this prevents us from using the 2KHz IMU mode.

Here is a modified version of your example ImuReader.cpp:
```
#include <iostream>
#include <mutex>
#include <libobsensor/ObSensor.hpp>
#include "utils.hpp"
#include <map>

#define ESC 27
std::mutex printerMutex;
int        main(int argc, char **argv) try {
    // Print the SDK version number, the SDK version number is divided into major version number, minor version number and revision number
    std::cout << "SDK version: " << ob::Version::getMajor() << "." << ob::Version::getMinor() << "." << ob::Version::getPatch() << std::endl;

    // Create a Context.
    ob::Context ctx;

    // Query the list of connected devices
    auto devList = ctx.queryDeviceList();

    if(devList->deviceCount() == 0) {
        std::cerr << "Device not found!" << std::endl;
        return -1;
    }

    // Create a device, 0 represents the index of the first device
    auto                        dev         = devList->getDevice(0);
    std::shared_ptr<ob::Sensor> gyroSensor  = nullptr;
    std::map<uint64_t, uint64_t> gyroSamples; // <timestamp, index>
    try {
        // Get Gyroscope Sensor
        gyroSensor = dev->getSensorList()->getSensor(OB_SENSOR_GYRO);
        if(gyroSensor) {
            // Get configuration list
            auto profiles = gyroSensor->getStreamProfileList();
            // Select the first profile to open stream
            auto profile = profiles->getGyroStreamProfile(OBGyroFullScaleRange::OB_GYRO_FS_1000dps, OBGyroSampleRate::OB_SAMPLE_RATE_200_HZ);

            gyroSensor->start(profile, [&](std::shared_ptr<ob::Frame> frame) {
                std::unique_lock<std::mutex> lk(printerMutex);
                auto                         timeStamp = frame->timeStampUs();
                auto                         index     = frame->index();
                auto                         gyroFrame = frame->as<ob::GyroFrame>();

                if(gyroFrame) {
                    auto it = gyroSamples.find(timeStamp);
                    if (it != gyroSamples.end()) {
                        std::cout << "Duplicate gyro sample! index0=" << std::to_string(it->second) << ", index1=" << std::to_string(index) << ", timestamp=" << std::to_string(timeStamp) << std::endl;
                    } else {
                        gyroSamples.insert(std::make_pair(timeStamp, index));
                    }
                }
            });
        }
        else {
            std::cout << "get gyro Sensor failed ! " << std::endl;
        }
    }
    catch(ob::Error &e) {
        std::cerr << "current device is not support imu!" << std::endl;
        exit(EXIT_FAILURE);
    }

    std::cout << "Press ESC to exit! " << std::endl;

    while(true) {
        // Get the value of the pressed key, if it is the ESC key, exit the program
        int key = getch();
        if(key == ESC)
            break;
    }

    // turn off the flow
    if(gyroSensor) {
        gyroSensor->stop();
    }
    return 0;
}
catch(ob::Error &e) {
    std::cerr << "function:" << e.getName() << "\nargs:" << e.getArgs() << "\nmessage:" << e.getMessage() << "\ntype:" << e.getExceptionType() << std::endl;
    exit(EXIT_FAILURE);
}
```

This gives me warning like:
`Duplicate gyro sample! index0=163, index1=164, timestamp=25591000`

zhonghong322, 2023-12-23:
1. Also, it would be great if you could fix the issue where IMU timestamps have 1ms accuracy; this prevents us from using the 2KHz IMU mode.

If there are further firmware updates planned, we will address and resolve this issue.

2. Duplicate gyro sample! index0=163, index1=164, timestamp=25591000  

I will test and reproduce the issue, and then provide you with a response

zhonghong322, 2024-01-17:
We have already fixed this issue. Please upgrade to firmware version 1.2.8 （https://github.com/orbbec/OrbbecFirmware/releases/tag/Femto-Mega-Firmware）and test it using OrbbecSDK version 1.9.3.（https://github.com/orbbec/OrbbecSDK/releases/tag/v1.9.3）

closed 2024-01-17 by kaatrasa