orbbec/OrbbecSDK #41: Femto Bolt timestamp reset occasionally not working with no errors

tim-depthkit, 2023-12-22:
I just ran into an issue when using 3x Femto Bolt cameras where two of them stopped responding to requests to reset the timestamps.

Power cycling the sensors resolved the issue, but it was quite confusing as the SDK reported no errors, but the timestamps were never reset.

This resulted in the entire system being unable to be synchronized, as the timestamps were so far apart.

zhonghong322, 2024-01-03:
* First, use the following interface to enable timestamp reset
void setTimestampResetConfig(const OBDeviceTimestampResetConfig &config);

Method of invocation:：
OBDeviceTimestampResetConfig resetConfig;
        resetConfig.enable = true;
        resetConfig.timestamp_reset_delay_us = 0;
    
        resetConfig.timestamp_reset_signal_output_enable = true;

        device->setTimestampResetConfig(resetConfig);

* timestamp reset
    (1) If resetting the timestamp through software, use the following interface (typically, when multiple devices are synchronized, it is sufficient to send the reset command to the master device, 
	which will automatically trigger the timestamp reset signal through TIMER_SYNC_OUT to the slave devices):
	
 void timestampReset();
 
Method of invocation:
device->timestampReset();

    (2) If resetting the timestamp through a hardware signal, use an external trigger signal to reset it.
        

*Notes：
The hardware timestamp reset feature and software clock synchronization feature (context::enableMultiDeviceSync() or context::enableDeviceClockSync()) are mutually exclusive for Femto Mega and Femto Bolt.

(1)To enable hardware timestamp reset:
resetConfig.enable = true;   //Indicates the use of the hardware timestamp reset feature

(2)To disable hardware timestamp reset and use software clock synchronization during multi-device synchronization: (Please refer to MultiDeviceSync.cpp)
resetConfig.enable = false;   //Disables the hardware timestamp reset feature, allowing the use of software clock synchronization during multi-device synchronization

tim-depthkit, 2024-01-03:
@zhonghong322 I am already performing the above steps to enable the timestamp reset functionality as well as calling timestampReset() on the device. This works successfully ~99% of the time, however I am reporting an issue or bug that it has occasionally not worked, and the device required a reboot in order for it to begin working again.
Note that this issue was encountered using firmware 1.0.6, I have not yet seen it happen on 1.0.9, but I have not done extensive testing with the new firmware yet.
I request your team look into potential causes within the device firmware code.