orbbec/OrbbecSDK #45: Femto Bolt sensors frequently appearing connected as USB2.1 while using USB 3.1 ports

tim-depthkit, 2024-01-05:
On Windows, after rebooting sensors, sometimes (quite often in fact) they will show up with USB2.1 connections when they become reconnected. This requires physically disconnecting and reconnecting the sensors, power cycling, or sometimes even rebooting the host PC in order to resolve the issue.

The issue occurs with both v1.0.6 and v1.0.9 firmware

xcy2011sky, 2024-01-06:
When a USB 3 device is enumerated as USB 2, it's often an issue related to impedance matching of the cable. Are you using the official cable provided with our products, or is it a custom cable that you have arranged? If it's a custom cable, could you please specify the length of the cable? Additionally, it would be helpful if you could provide some images or specifications of the cable. This will allow us to assess the situation more accurately.

ScatterCory, 2024-01-10:
@xcy2011sky I am working with Tim using these same sensors. We are using the USB cable supplied with the sensor, however we are mating that cable in-line with a [StarTech USB3AAEXT10M](https://www.startech.com/en-us/cables/usb3aaext10m) 10m Active Extension cable to connect the sensor to the host PC at a distance of up to 11 meters away - this total length of cable is necessary for our use case. This exact SKU of USB extension cable has worked without issue when using the Azure Kinect in the same context.

xcy2011sky, 2024-01-16:
@ScatterCory  Thank you for your message. We will acquire some wiring to recreate the issue. After successfully replicating it, we will focus on resolving this issue in the upcoming version improvement.

ChuckM90, 2024-01-16:
> On Windows, after rebooting sensors, sometimes (quite often in fact) they will show up with USB2.1 connections when they become reconnected. This requires physically disconnecting and reconnecting [...]

I have the same problem connecting USB2.1/3.1 with a Femto Bolt using the supplied cable from the box.
I did a test using also another Femto Bolt, same cable, same pc, same USB 3.1 port, same firmware 1.0.9 but with one camera after I turn off and on the pc it gives me a usb 2.1 connection while the other always usb 3.1. It should be a hardware problem so I will send a return for repair.

obviousjim, 2024-01-17:
@xcy2011sky James here from Scatter. Thanks for taking action and testing those cables to see if you can make this connectivity issue more robust. It's certainly the leading cause of pain for us right now, in an otherwise very well built product.

Do any of the bug fixes in v1.9.3 SDK release address this connectivity issues? https://github.com/orbbec/OrbbecSDK/releases/tag/v1.9.3

Rudizzzz, 2024-01-26:
I have the same problem as @ChuckM90.
I'm using the official USB cable provided in the box and some Bolt runs on 2.1 and others on 3.1.
The ones that run on 2.1 after some disconnection / reconnection start on 3.1.
The others run always on 3.1 even after many disconnection / reconnection.
The PC usb port is the same for every Bolt tested.
FW: 1.0.9.
SDK: 1.8.3.
Let me know if you need any information and please how to address this issue.

dj-af, 2024-09-26:
@xcy2011sky We are experiencing the same issues as described in this thread, but we are using the cables provided in the box. In our experience a camera suddenly switches to be a USB2.1 device after having run as a 3.1 device for a long time. Sometimes we can "fix" it by switching the USB cable to a different USB port, sometimes by just unplugging and re-plugging the USB cable. 

However, once our devices are out with our customers, it becomes a bigger problem to perform these manual fixes. Here is an example of a customer's camera that suddenly changed from USB3.1 to USB2.1 a few days ago:

[image]

Please advise what can be done to prevent this issue from happening.

Trekopep, 2024-10-04:
> Femto Bolt long-distance transmission, need to find a matching usb cable; Different usb cables have impedance matching problems, resulting in probability recognition 2.0; Refer to the following links, https://newnex.com/documents/Orbbec_Newnex_InteroperabilityTest.pdf

Can I assume that the listed cables for Femto Bolt 2 would also work for the Femto Mega? I've been using this [Insignia™ - 3.3’ USB to USB-C 3.2 Gen 2 Superspeed+ 10Gbps Cable](https://www.bestbuy.com/site/insignia-3-3-usb-to-usb-c-3-2-gen-2-superspeed-10gbps-cable-black/6419159.p?skuId=6419159), which tends to usually work on its own, but if I pair it with this [Amazon Basics 3 Meter USB-A 3.0 Extension Cable](https://a.co/d/9ofzFKK) it frequently switches between showing USB 3 and USB 2, and if I pair it with this other [20 ft USB 3.0 Extension Cable](https://a.co/d/c9hms2G) it seems to very rarely connect at all.

Also, I recognize that I should probably be able to use an ethernet connection to send data from the Femto Mega, but it seems like maybe that's not supported because I'm using the K4A-wrapper to have access to Azure Kinect's body tracking SDK?

Here's a bit from the Orbbec Viewer logs when the Amazon Basics cable kept disconnecting including a crash at the end, in case it's useful:
<details><summary>Logs Code Block</summary>

```
...
2024-10-03 21:44:59.501 ERROR [20072] [PropertyManager::syncProperty@478] init property failed: 2015@Power Line Frequency, Device is deteched.
2024-10-03 21:44:59.501 ERROR [20072] [PropertyManager::write@531] write property failed: 2001@Color Sensor Exposure, ob::Device::setIntProperty, exceptionType: 6, message: Request failed, statusCode: 2, msg: send control transfer failed!
2024-10-03 21:44:59.501 ERROR [20072] [PropertyManager::read@586] write property failed: 2001@Color Sensor Exposure, ob::Device::getIntProperty, exceptionType: 6, message: Request failed, statusCode: 2, msg: send control transfer failed!
2024-10-03 21:44:59.502 ERROR [20072] [PropertyManager::read@586] write property failed: 2005@UVC image brightness, ob::Device::getIntProperty, exceptionType: 8, message: Device is deteched.
2024-10-03 21:44:59.502 ERROR [20072] [PropertyManager::read@586] write property failed: 2002@Color Sensor Gain, ob::Device::getIntProperty, exceptionType: 8, message: Device is deteched.
2024-10-03 21:44:59.502 ERROR [20072] [PropertyManager::read@586] write property failed: 2004@UVC image white balance setting, ob::Device::getIntProperty, exceptionType: 8, message: Device is deteched.
2024-10-03 21:44:59.502 INFO  [20072] [OBSensorExt::syncPropertiesFunc@1122] OBSensorExt sync all properties finish.
2024-10-03 21:45:02.118 DEBUG [704] [OBSensorServer::{ctor}::<lambda_15c7ea970e758fad25b736428dfbc7b6>::operator ()@66] device added:Femto Mega-uid:15589d510
2024-10-03 21:45:02.375 INFO  [10556] [OBSensorExt::setCurrentStreamProfile@347] setStreamProfile@OB_STREAM_DEPTH: 640x576 15fps Y16
2024-10-03 21:45:02.376 INFO  [10556] [OBSensorExt::setCurrentStreamProfile@347] setStreamProfile@OB_STREAM_IR: 640x576 15fps Y16
2024-10-03 21:45:02.378 INFO  [10556] [OBSensorExt::setCurrentStreamProfile@347] setStreamProfile@OB_STREAM_COLOR: 1920x1080 15fps MJPEG
2024-10-03 21:45:04.315 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 14@Mirror Depth Stream -> 0, range: 0 - 1, default: 0
2024-10-03 21:45:04.315 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 15@Flip Depth Stream -> 0, range: 0 - 1, default: 0
2024-10-03 21:45:04.315 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 18@IR Mirror -> 0, range: 0 - 1, default: 0
2024-10-03 21:45:04.316 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 19@IR Flip -> 0, range: 0 - 1, default: 0
2024-10-03 21:45:04.316 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 24@Depth soft filter -> 0, range: 0 - 1, default: 0
2024-10-03 21:45:04.316 INFO  [11812] [PropertyManager::syncProperty@448] sync property: 40@Max Diff -> 300, range: 0 - 10000, default: 300
2024-10-03 21:45:04.316 INFO  [11812] [PropertyManager::syncProperty@448] sync property: 41@Max Speckle Size -> 25, range: 0 - 10000, default: 25
2024-10-03 21:45:04.322 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 42@Hardware D2C -> 0, range: 0 - 1, default: 1
2024-10-03 21:45:04.325 INFO  [11812] [PropertyManager::syncProperty@448] sync property: 43@color frame timestamp offset -> 0, range: 0 - 0, default: 0
2024-10-03 21:45:04.328 INFO  [11812] [PropertyManager::syncProperty@448] sync property: 62@Fan work mode -> 1, range: 0 - 1, default: 0
2024-10-03 21:45:04.331 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 81@Mirror RGB Stream -> 0, range: 0 - 1, default: 0
2024-10-03 21:45:04.333 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 82@Flip RGB Stream -> 0, range: 0 - 1, default: 0
2024-10-03 21:45:04.335 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 83@Light -> 1, range: 0 - 1, default: 1
2024-10-03 21:45:04.337 INFO  [11812] [PropertyManager::syncProperty@441] sync property: 89@Heartbeat Detection Enable -> 0, range: 0 - 1, default: 0
2024-10-03 21:45:04.338 INFO  [11812] [PropertyManager::syncProperty@448] sync property: 97@Communication Type -> 0, range: 0 - 1, default: 0
2024-10-03 21:45:04.340 INFO  [11812] [PropertyManager::syncProperty@448] sync property: 98@IR Mode -> 0, range: 0 - 1, default: 0
[... cut ...]
2024-10-03 21:45:20.591 ERROR [10556] [OBSensorExt::initProfileList@294] Device is disconnect!
2024-10-03 21:45:20.591 ERROR [10556] [OBSensorExt::initProfileList@294] Device is disconnect!
2024-10-03 21:45:21.625 WARN  [10556] [OBImuSensorExt::addSensor@368] add imu sensor failed!
2024-10-03 21:45:21.626 ERROR [10556] [D2CPipeline::updateColorProfileList@352] Device is disconnect!
2024-10-03 21:45:21.627 ERROR [10556] [D2CPipeline::updateColorProfileList@352] Device is disconnect!
```
</details>

ChuckM90, 2024-10-24:
After a hundred Femto Bolt cameras used on our machinery I can give you some feedback.
After updating to firmware 1.1.0 only 2 had the same problem after the update. Then restarting the camera the problem never recurred again. I'm using the cable in the package and connected to a USB 3.1 port

Hyeonho-Lee, 2025-01-31:
We have been operating multiple Femto Bolt devices, and we encountered an issue where a specific desktop PC kept recognizing the device as USB 2.1.

We attempted various solutions, including replacing the cable, updating the camera firmware, updating the Orbbec SDK, and even replacing the motherboard, but the issue persisted.

Eventually, we successfully connected the device to a USB 3.1 port by disabling the "Legacy USB Support" option in the BIOS settings of the affected PC.

funatsufumiya, 2025-01-31:
I faced similar problem when extending usb cable.

We usually used this one (5Gbps USB 3.1 extension cable).
https://newnex.com/usb-3-1-type-c-active-cable.php

But we found Femto Bolt needs this one (10Gbps USB 3.2 extension cable).
https://newnex.com/firenex-ulink-10g.php

ChuckM90, 2025-01-31:
> We have been operating multiple Femto Bolt devices, and we encountered an issue where a specific desktop PC kept recognizing the device as USB 2.1.
>
> We attempted various solutions, including [...]

It occurred to me with this workaround of yours regarding USB, that it is very important that "fast boot"in "Power energy plan" is disabled on Windows PCs. I have had experience with other USB devices that no longer worked on the second boot of the PC if enabled. It is truly a harmful option which instead of being disabled by default is instead enabled (congratulations Microsoft)

sjw9912, 2025-04-20:
I have the same problem, even with the cable that came in the package.
I'm so tired of USB recognition issues. I want a refund for my Femto bolt