orbbec/OrbbecSDK #53: depthengine.so name collision with Kinect Azure

flugi, 2024-02-17:
Is there a way to use Orbbec and Kinect Azure in the same time? The libdepthengine.so.2.0 file is opened from both SDKs, and the file is not the same, Orbbec malfunctions if I use k4a version of libdepthengince.so.2.0, so it seems I have to choose which one I want to use and use only the corresponding shared object file. 

Is there a best practice to how to use both devices in the same program in the same time?

thundercarrot, 2024-02-21:
Azure Kinect SDK is open source. Maybe you can change the name of the library that is dynamically loaded?

natelowry, 2024-02-21:
@thundercarrot unfortunately I think the depthengine library is not open source :/ 

[DepthEngine readme](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/depthengine.md)

hzcyf, 2024-02-22:
We have provided a sample code to deal with this issue, please refer to: https://github.com/orbbec/K4ALoader/tree/feature/loader_c%2B%2B

hzcyf, 2024-02-22:
> We have provided a sample code to deal with this issue, please refer to: https://github.com/orbbec/K4ALoader/tree/feature/loader_c%2B%2B

As the sample code is an early version and hard to read, I recommend you to read this directly:
https://github.com/orbbec/K4ALoader/blob/77f20c00c98740bb92a9922e53ad9cc3de46f185/main.cpp#L926
[image]

hzcyf, 2024-02-27:
Close as resolved.

hzcyf, 2024-02-27:
Close as resolved.

closed 2024-02-27 by hzcyf