orbbec/OrbbecSDK #57: RGB image exposure time and gain

kaatrasa, 2024-02-20:
Hey,

It would be very useful for us to have access to image exposure time and gain values. Currently, we can try to query the exposure time and gain from the device like this:
```
device->getIntProperty(OB_PROP_COLOR_EXPOSURE_INT)
device->getIntProperty(OB_PROP_COLOR_GAIN_INT)
```
and these work when we are not using auto exposure. (One issue is that the exposure time unit changes between devices; Astra2 uses microseconds, Femto Mega uses 0.1ms).

However, with AE enabled, these values are not updated at all and so we cannot even get rough values for exposure time and gain. Ideally, we would like to get image specific values, i.e. could you consider adding these outputs to OrbbecSDK when you implement this support in Orbbec K4A Wrapper? [k4a_image_get_exposure_usec](https://github.com/orbbec/OrbbecSDK-K4A-Wrapper/blob/ffb09cb6c08e08047e8a086529f8156dd3a884d6/include/k4a/k4a.h#L996C21-L996C48), [k4a_image_get_iso_speed](https://github.com/orbbec/OrbbecSDK-K4A-Wrapper/blob/ffb09cb6c08e08047e8a086529f8156dd3a884d6/include/k4a/k4a.h#L1047)

zhonghong322, 2024-02-27:
What applications do you use the parameters of exposure and ISO speed obtained for each frame for?

kaatrasa, 2024-02-27:
Couple of examples (related to VISLAM)

1. Implementing custom auto-exposure; we would like to minimize exposure time to avoid motion blur.
2. Getting mid point of image exposure; in earlier issue you mentioned the Orbbec images have timestamp at start of exposure and ideally we would like to know the mid point. 
3. Motion blur compensation

biggiantpigeon, 2025-05-09:
Has this issue any progress now？I notice it's been more than a year.
My suggestion is to send exposure time and gain through ros topic, like in /camera/color/image_raw, dunno if that's possible.