orbbec/OrbbecSDK #59: Frame index abnormal warnings on Femto Mega with ethernet

flugi, 2024-02-26:
According to the warnings, there are dropped frames ( example: #206 [WARN] Frame index abnormal, prevFrameIndex=89557, curFrameIndex=89559) when used over ethernet. As far as I can see it, no bottlenecks are present, the OrbberViewer utilizes the CPU ~50% of a single core, the bandwidth is ~27Mbyte/s after a gigabit handshake, so pretty far from being bottleneck. No ethernet messages in dmesg, so I assume on hardware level it works as it should.

Is this some false error report? Or should I play around priorities / formats / etc?

zhonghong322, 2024-02-27:
What is the format and frame rate of the stream you are opening, and what is the configuration of your PC?

flugi, 2024-02-27:
depth+color enabled, 640x576, 30 fps, 1080p color MJPEG

CPU: AMD 4800H, the full config is a Tuxedo Pulse 14 Gen1 laptop. Running as user. The ~50% of single core report is in htop, view as threads, there are more active threads, and the max is ~50% of a core.

zhonghong322, 2024-02-28:
Are you using OrbbecViewer for testing? What is the frame rate displayed by OrbbecViewer? Is it occasional frame drops over the network or consistently low frame rate?

Have you tested the frame rate using OrbbecViewer via USB? Is the frame rate normal?

flugi, 2024-02-28:
Yes, I use OrbbecViewer for testing. In a more recent test I use 640x576 30fps, it shows 29.9 fps and 176 Mbit/s, which is confirmed using the OS network widget showing 22 MByte/s, and the frame drop occurs less, I have 8 frame drop events in ~90000 frame, so it seems this is a probabilistic phenomena. The previous test was also used 1080p MJPEG with 27MByte/s bandwidth and I had roughly 4x the frame drops. Using only the low resolution had no frame drops in half an hour testing. I will make some USB interface tests as well and also a different ethernet cable and get back to you.

flugi, 2024-02-28:
other cable also drop frames for me. If I increase the bandwidth requirements, the frame dropping also increases. IR+Depth 640x576 + 1080p MJPEG drops every ~1000th frame, 10x the depth only mode.