orbbec/OrbbecSDK #72: No saved gemini E parameters

June-Jo, 2024-03-20:
I want to know intrinsic parameters of Gemini E in Ubuntu 22.04.
I tried 2 ways to know them.

First, I tried to use API like below code.

int main(int, char**){
    ob::Pipeline pipe;
    auto device = pipe.getDevice();
    auto cameraParamList = device->getCalibrationCameraParamList();
    for(int i = 0; i < cameraParamList->count(); i++) {
    OBCameraParam obParam = cameraParamList->getCameraParam(i);
    std::cout << "fx:" << obParam.depthIntrinsic.fx << " fy: " << obParam.depthIntrinsic.fy << " cx:" << obParam.depthIntrinsic.cx
    << " cy: " << obParam.depthIntrinsic.cy
    << " width: " << obParam.depthIntrinsic.width << " height:" << obParam.depthIntrinsic.height << std::endl;}
}

I got no the no message and I noticed the cameraParamList->count() is 0.

Second, I started "OBCommonUsages" in SDK and make it print parameters.
And it says, all parameters are 0 like below.

Input command:  p
depthIntrinsic fx:0, fy0, cx:0, cy:0 ,width:0, height:0
rgbIntrinsic fx:0, fy0, cx:0, cy:0, width:0, height:0
depthDistortion k1:0, k2:0, k3:0, k4:0, k5:0, k6:0, p1:0, p2:0
rgbDistortion k1:0, k2:0, k3:0, k4:0, k5:0, k6:0, p1:0, p2:0
transform-rot: [0, 0, 0, 0, 0, 0, 0, 0, 0]
transform-trans: [ 0,  0,  0]

How can I know the intrinsic parameters of my Gemini E?

zhonghong322, 2024-03-23:
> I want to know intrinsic parameters of Gemini E in Ubuntu 22.04. I tried 2 ways to know them.
>
> First, I tried to use API like below code.
>
> int main(int, char**){ ob::Pipeline pipe; auto device = [...]
I tested it on Ubuntu 22.04, and it's working fine.
[image]

zhonghong322, 2024-03-23:
1 Obtain Camera Parameters through the Device
The intrinsic parameters obtained through the Device are the raw intrinsic parameters obtained directly from the device, without depth-to-color (D2C) . It is recommended for non-professionals to obtain parameters from the pipeline.
Use case: When only Depth is enabled and D2C alignment is not required, the intrinsic parameters can be obtained from the Device. The intrinsic parameters obtained from the Device are provided as a list, and the appropriate set of intrinsic parameters should be selected based on the resolution, matching the streaming resolution.

ob::Pipeline pipe;   
auto device = pipe.get device ();
auto cameraParamList = device->getCalibrationCameraParamList();
// Print the intrinsic parameters.
for(int i = 0; i < cameraParamList->count(); i++) {
OBCameraParam obParam = cameraParamList->getCameraParam (I);
    std::cout << "fx:" << obParam.depthIntrinsic.fx << " fy: " << obParam.depthIntrinsic.fy << " cx:" << obParam.depthIntrinsic.cx
              << " cy: " << obParam.depthIntrinsic.cy 
        << " width: " << obParam.depthIntrinsic.width << " height:" << obParam.depthIntrinsic.height << std::endl;
    
}

2 Obtain Camera Parameters through Pipeline
To obtain the camera's intrinsic parameters through the Pipeline, enable the Depth stream and retrieve the intrinsic parameters for Depth. Refer to the DepthViewer sample for obtaining the Depth stream. Additionally, obtain the intrinsic parameters after depth-to-color (D2C) alignment by referring to the SyncAlignViewer sample, which demonstrates obtaining both the Depth and Color streams.

Note: To obtain camera parameters by Pipiline, you must obtain them after the stream is turned on.

ob::Pipeline pipe;
pipe.start(config);
auto camera_param = pipe.getCameraParam();

zhonghong322, 2024-03-23:
> I want to know intrinsic parameters of Gemini E in Ubuntu 22.04. I tried 2 ways to know them.
>
> First, I tried to use API like below code.
>
> int main(int, char**){ ob::Pipeline pipe; auto device = [...]

Is it that your device parameters are lost or not calibrated?  Please test by obtaining parameters through the pipeline.

June-Jo, 2024-03-28:
@zhonghong322 Wow, I tried again by using "OBCommonUsages" bin file, and I succeeded to get parameters by getting parameter command after streaming command.

[image]

When OBCommonUsages is executed, the rgb streaming window comes up so I miscatched the streaming is started automatically.
Thank you for your help!
Have a nice day.

June-Jo, 2024-03-28:
@zhonghong322 I noticed the parameters are not normal and the values are changing when I try to get them.
Can you give me an advice to get the parameters from device?

I tried like below but still the result is not normal.

#include <iostream>
#include "libobsensor/ObSensor.hpp"

int main(int, char**){
    ob::Pipeline pipe;
    auto device = pipe.getDevice();
    std::shared_ptr<ob::Config> config = std::make_shared<ob::Config>();
    std::shared_ptr<ob::VideoStreamProfile> colorProfile = nullptr;

    try {
        auto profiles = pipe.getStreamProfileList(OB_SENSOR_COLOR);
        try {
            colorProfile = profiles->getVideoStreamProfile(640, 480, OB_FORMAT_RGB, 30);
        }
        catch(ob::Error &e) {
            colorProfile = std::const_pointer_cast<ob::StreamProfile>(profiles->getProfile(OB_PROFILE_DEFAULT))->as<ob::VideoStreamProfile>();
        }
        config->enableStream(colorProfile);
    }
    catch(ob::Error &e) {
        std::cerr << "Current device is not support color sensor!" << std::endl;
        exit(EXIT_FAILURE);
    }

    pipe.start(config);

    auto camera_param = pipe.getCameraParam();

    std::cout << "fx:" << camera_param.rgbIntrinsic.fx << " fy: " << camera_param.rgbIntrinsic.fy << " cx:" << camera_param.rgbIntrinsic.cx
        << " cy: " << camera_param.rgbIntrinsic.cy
        << " width: " << camera_param.rgbIntrinsic.width << " height:" << camera_param.rgbIntrinsic.height << std::endl;

    auto cameraParamList = device->getCalibrationCameraParamList();
    
    std::cout << "cameraParamList->count()\n" << cameraParamList->count() << std::endl;
    for(int i = 0; i < cameraParamList->count(); i++) {
        OBCameraParam obParam = cameraParamList->getCameraParam(i);
        std::cout << "fx:" << obParam.rgbIntrinsic.fx << " fy: " << obParam.rgbIntrinsic.fy << " cx:" << obParam.rgbIntrinsic.cx
        << " cy: " << obParam.rgbIntrinsic.cy
        << " width: " << obParam.rgbIntrinsic.width << " height:" << obParam.rgbIntrinsic.height << std::endl;
    }
}

[image]

June-Jo, 2024-04-12:
I got the parameters using viewer.

closed 2024-04-12 by June-Jo

mightypanda, 2024-10-10:
Hello @June-Jo , @zhonghong322 

I cannot retrieve Depth Distortion for a Gemini2 camera, they are always zero, while RGB distortion is provided.
This happens both in viewer and using sdk.

I need those to convert a single depth point to a 3d point.

[image]