orbbec/OrbbecSDK #74: Femto Bolt: Crash after "Process frame failed!"

puzzlepaint, 2024-03-29:
With Femto Bolt cameras, aside from [this very frequent crash](https://github.com/orbbec/OrbbecSDK/issues/73), I am encountering another occasional crash after starting a camera on Linux. Here is a log from one such occurrence:

```
[...]
[03/28 23:12:26.521777][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:26.522123][info][37527][Pipeline.cpp:239] Try to start streams!
[03/28 23:12:26.522144][info][37527][VideoSensor.cpp:663] start OB_SENSOR_COLOR stream with profile: {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 2560, height: 1440, fps: 30}
[03/28 23:12:26.523633][info][37527][VideoSensor.cpp:663] start OB_SENSOR_DEPTH stream with profile: {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 30}
[03/28 23:12:26.523637][info][37527][MSDEConverterDevice.cpp:544] Start real profile,width:7680 height:434
[03/28 23:12:26.868000][info][37543][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:492272
[03/28 23:12:26.868023][info][37543][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[03/28 23:12:26.970372][info][37543][MSDEConverterDevice.cpp:113] Depth engine init succeed!
[03/28 23:12:26.970535][info][37527][VideoSensor.cpp:663] start OB_SENSOR_IR stream with profile: {type: OB_STREAM_IR, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 30}
[03/28 23:12:26.970548][info][37527][Pipeline.cpp:252] Start streams done!
[03/28 23:12:26.970554][info][37527][Pipeline.cpp:235] Pipeline start done!
[03/28 23:12:27.400317][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:27.602040][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:27.804139][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:28.005755][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:28.207605][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:28.409200][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:28.610953][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:28.813537][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:29.015372][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:29.217262][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:29.419256][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:29.622991][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=2ms
[03/28 23:12:29.825306][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:30.027110][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:30.228845][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:30.430660][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:30.632443][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:30.834166][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:31.035946][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:31.237759][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:31.439593][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:31.642483][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=1ms
[03/28 23:12:31.844234][info][37535][DeviceManager.cpp:249] dev-uid@0x6-2-8, update device time succeeded! round-trip-time=0ms
[03/28 23:12:31.914304][error][37543][MSDEConverterDevice.cpp:322] Process frame failed!

Thread 35 "OrbbecThread" received signal SIGSEGV, Segmentation fault.
```

I was testing frequent timer updates here which is why there are many "update device time succeeded" messages, but I don't think this is necessarily related to the crash. 

The call stack for the crash is somewhere within libOrbbecSDK.so.1.9:
```
(gdb) bt
#0  0x00007ffff33cefed in ?? () from /home/.../third_party/OrbbecSDK/lib/linux_x64/libOrbbecSDK.so.1.9
#1  0x00007ffff33d1783 in ?? () from /home/.../third_party/OrbbecSDK/lib/linux_x64/libOrbbecSDK.so.1.9
#2  0x00007ffff2edc3ec in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff2a94ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#4  0x00007ffff2b26850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

Without having counted it, I think that this crash roughly happens every 20th time when starting a camera.

I am on Linux, with the Orbbec SDK V1.9.5, firmware version 1.1.0 for the Femto Bolt, and so far using the original USB cable without an extension.

zhonghong322, 2024-04-07:
I see you mentioned two crash issues. Could you provide me with the specifications of your computer, such as CPU and GPU models? We'd like to try reproducing the issue.

puzzlepaint, 2024-04-07:
The specifications of the PC on which I observed the crashes are:
```
CPU: AMD Ryzen 9 7950X 16-Core Processor
RAM: 64 GB
GPU: NVIDIA GeForce RTX 4090
OS: Ubuntu 22.04.4 LTS
```