orbbec/OrbbecSDK #76: OrbbecViewer_v1.9.5 fails to startup when (10) Femto Bolt sensors are connected to the host PC

ScatterCory, 2024-04-02:
When 10 Femto Bolt sensors are connected to my computer simultaneously, OrbbecViewer fails to fully startup, and instead remains unresponsive with just the console and a white screen.

OrbbecViewer v1.8.3 does not have this issue, and is able to startup with all 10 sensors connected. Additionally, our Depthkit application for Windows is able to control and capture streams from all 10 sensors simultaneously.

I have experienced this issue with all sensors running firmware v1.1.0 , and also with all sensors running firmware v1.0.9.

I have attached the contents of the console here: 
[OrbbecViewer_v1.9.5_202403051246_win_x64_release_10sensorErrorLog.txt](https://github.com/orbbec/OrbbecSDK/files/14842551/OrbbecViewer_v1.9.5_202403051246_win_x64_release_10sensorErrorLog.txt)

[image]

zhonghong322, 2024-04-24:
You can bypass the issue by setting the configuration file to AutoConnect=false. We will address it in future updates.
[image]

ScatterCory, 2024-05-01:
@zhonghong322 This change the the config.ini file appears to have resolved the issue, and I can launch Orbbec Viewer v1.9.5 with all 10 sensors connected. Thank you!

closed 2024-05-01 by ScatterCory