orbbec/OrbbecSDK #79: Does Femto Bolt support Ubuntu 22.04 ?

omarzandona, 2024-04-11:
Hello Orbbec Team,

From the documentation, I have seen that Femto Bolt supports Ubuntu version 18.04 and later. Indeed, I managed to get it working on both amd64 and arm64 architectures (by compiling examples from the OrbbecSDK repository) with Ubuntu 20.04, but I'm unable to get it to work with Ubuntu 22.04 and later versions (both on amd64 and ARM64-Raspberry Pi 4B - and Ubuntu with versions higher than 20.04).

Even using a Docker with Ubuntu 20.04, I can't get it to work on Ubuntu 22.04 and later hosts.

On every platform, I get this error (arm64 architecture in this case):

```
june@u22:~ $ LD_LIBRARY_PATH=/home/june/OrbbecSDK/lib/arm64 ./OBDepthViewer 
[04/10 16:33:49.285548][info][38766][Context.cpp:69] Context created with config: default config!
[04/10 16:33:49.285572][info][38766][Context.cpp:74] Context work_dir=/home/june/OrbbecSDK/build/bin
[04/10 16:33:49.285579][info][38766][Context.cpp:77]    - SDK version: 1.9.5
[04/10 16:33:49.285591][info][38766][Context.cpp:78]    - SDK stage version: main
[04/10 16:33:49.285606][info][38766][Context.cpp:82] get config EnumerateNetDevice:false
[04/10 16:33:49.285620][info][38766][LinuxPal.cpp:38] createObPal: create LinuxPal!
[04/10 16:33:49.392381][info][38766][LinuxPal.cpp:112] Create PollingDeviceWatcher!
[04/10 16:33:49.392424][info][38766][DeviceManager.cpp:15] Current found device(s): (1)
[04/10 16:33:49.392436][info][38766][DeviceManager.cpp:24]  - Name: Femto Bolt, PID: 0x066b, SN/ID: CL8FC3100G7, Connection: USB3.1
[04/10 16:33:49.392452][info][38766][Pipeline.cpp:15] Try to create pipeline with default device.
[04/10 16:33:49.392529][info][38766][FemtoBoltUvcDevice.cpp:23] FemtoBoltUvcDevice init ...
[04/10 16:33:49.392619][info][38766][FemtoBoltUvcDevice.cpp:120] Create command start!
[04/10 16:33:49.396012][info][38766][MSDEConverterDevice.cpp:721] Succeed to load depth engine plugin
[04/10 16:33:49.604886][info][38766][FemtoBoltUvcDevice.cpp:271] Create command done!
[04/10 16:33:49.604911][info][38766][FemtoBoltUvcDevice.cpp:431] init sensor map start!
[04/10 16:33:49.604943][info][38766][FemtoBoltUvcDevice.cpp:458] init sensor map done!
[04/10 16:33:49.605711][info][38766][AbstractDevice.cpp:124]    - Firmware version: 1.0.9
[04/10 16:33:49.605855][info][38766][FemtoBoltUvcDevice.cpp:275] Init depth process param start!
[04/10 16:33:49.616655][info][38766][MSDEConverterDevice.cpp:772] got nvram data succeed.
[04/10 16:33:49.616692][info][38766][ObUvcDevice.cpp:112] endpoint:130
[04/10 16:33:49.616776][info][38766][ObUvcDevice.cpp:118] libusb_clear_halt done, endpoint:130
[04/10 16:33:50.060722][info][38766][FemtoBoltUvcDevice.cpp:402] setNvramDataStreamStopFunc succeed
[04/10 16:33:50.060834][info][38766][FemtoBoltUvcDevice.cpp:427] Init depth process param done!
[04/10 16:33:50.062184][info][38766][FemtoBoltUvcDevice.cpp:38] FemtoBoltUvcDevice init done!
[04/10 16:33:50.062216][info][38766][DeviceManager.cpp:157] Device created successfully! Name: Femto Bolt, PID: 0x066b, SN/ID: CL8FC3100G7
[04/10 16:33:50.062293][info][38766][Pipeline.cpp:44] Pipeline created with device: {name: Femto Bolt, sn: CL8FC3100G7}, @0xAAAADEC6AF40
[04/10 16:33:50.062519][info][38766][FemtoBoltUvcDevice.cpp:554] Depth sensor has been created!
[04/10 16:33:50.062680][warning][38766][ObException.cpp:5] Invalid input, No matched video stream profile found!
[04/10 16:33:50.064161][info][38766][Pipeline.cpp:239] Try to start streams!
[04/10 16:33:50.064255][info][38766][VideoSensor.cpp:663] start OB_SENSOR_DEPTH stream with profile: {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 30}
[04/10 16:33:50.064269][info][38766][MSDEConverterDevice.cpp:544] Start real profile,width:7680 height:434
[04/10 16:33:50.066761][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:50.066797][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:50.291359][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:50.291467][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:50.291481][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:50.419066][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:50.419263][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:50.419298][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:50.546213][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:50.546494][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:50.546572][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:50.673891][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:50.673999][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:50.674009][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:50.804816][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:50.804925][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:50.804935][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:50.935567][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:50.935674][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:50.935685][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:51.065487][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:51.065600][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:51.065611][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:51.192710][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:51.192925][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:51.193019][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:51.317203][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:51.317403][info][38782][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:494408
[04/10 16:33:51.317424][info][38782][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[04/10 16:33:51.445483][error][38782][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[04/10 16:33:51.445803][info][38766][Pipeline.cpp:252] Start streams done!
[04/10 16:33:51.445835][info][38766][Pipeline.cpp:235] Pipeline start done!
QSocketNotifier: Can only be used with threads started with QThread
[04/10 16:33:51.547534][warning][38766][Pipeline.cpp:328] Wait for frame timeout, you can try to increase the wait time! current timeout=100
[04/10 16:33:54.553466][warning][38766][Pipeline.cpp:328] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**30 logs in 3005ms**]
[04/10 16:34:00.564772][warning][38766][Pipeline.cpp:328] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**60 logs in 6011ms**]
[04/10 16:34:12.575744][warning][38766][Pipeline.cpp:328] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**120 logs in 12010ms**]
[04/10 16:34:36.609110][warning][38766][Pipeline.cpp:328] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**240 logs in 24033ms**]
```

Has anyone else encountered the same issue? If so, how did you resolve it?

NOTE: Using Femto Mega, I didn't encounter any issues on Ubuntu 22.04 and later versions.

Thank you for the support.

Best regards,
Omar

jian-dong, 2024-04-24:
Hello Omar,

Thank you for your detailed description of the issues you are encountering. It appears that the problem may stem from the NVIDIA GPU drivers not being properly loaded within your Docker environment. To ensure that your system utilizes the NVIDIA GPU correctly, you need to enable GPU support in Docker.

Please follow the NVIDIA Container Toolkit installation guide available here: [NVIDIA Container Toolkit Installation Guide](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html). This guide will help you set up the Docker environment to correctly interact with the NVIDIA GPUs.

If you have any further questions or encounter additional issues after following the guide, please feel free to reach out.

omarzandona, 2024-04-29:
Hello jian-dong,

Thank you for your prompt response and for providing insights.

I wanted to inform you that my system does not have an NVIDIA GPU installed. Consequently, it may not be feasible for me to proceed with the suggested solution.

Based on my past experience, I believe Femto Bolt can still function when connected to a system with only a CPU, albeit with potentially slower performance compared to systems equipped with an NVIDIA GPU. I've successfully used Femto Bolt on my old system running Ubuntu 18.04 without a GPU, and I didn't encounter any issues (with or without docker).

If you have any alternative suggestions or recommendations that are applicable to systems without NVIDIA GPUs, I would greatly appreciate your insights.

Thank you again for your assistance and understanding. I look forward to your response.

lirichardil, 2024-10-15:
Hi all,

I am experiencing similar. Currently, I am running Orbbec SDK on Ubuntu 22.02 on RPI 4. The color viewer seems fine, but when running the DeptchViewer(from the example), I got the same error message. The DepthViewer window will pop up but with no image displayed. 

I also installed OpenGL as many others has suggested, but no luck so far. 

Any potential resolution to this issue?

Thanks,
Richard

omarzandona, 2025-04-03:
Dear Yuxuanzhang

_excuse me, that I have enquiries how can you manage to operate Femto Bolt on Linux ARM64?_ 

I managed to get it working on both amd64 and arm64 (Raspberry Pi 4B) architectures with Ubuntu 20.04.
I compiled examples from the OrbbecSDK repository and used Orbbec Viewer to grab frames and acquires shorts video using both architectures.

While I was unable to get it to work with  Ubuntu 22.04 and later versions (both on amd64 and ARM64-Raspberry Pi 4B). See my first comment above.

_I use a VM on my MacBook, but it seems I cannot get access to the depth camera (with no monitoring window and data)
Additionally, the Femto Bolt official website also reveals that femto bolt cannot support arm linux, so I was wondering the potential solutions, thanks a lot._

It is not possible to effectively emulate Linux ARM64 on your MacBook using a VM, as the ARM64 architecture is different from the x86_64 architecture that MacBooks typically use.
To run on native ARM64 hardware you have to consider use, for example, a Raspberry Pi 4B.
Moreover, this [link](https://github.com/orbbec/OrbbecSDK?tab=readme-ov-file#platform-support) says that Femto Bolt is not supported by MacOS.
I am using this link for here you can find the compatibility and requirements. It seems that the one you mentioned is about converting Orbbec SDK to Azure Kinect Sensor SDK interfaces and could have different specs or requirements.

Best regards,
Omar Zandona