orbbec/OrbbecSDK #87: Unity Wrapper Support?

noemis84, 2024-05-15:
hey there,
I wrote an issue in the unity section [here](https://github.com/orbbec/OrbbecUnitySDK/issues/3)

Because there is no activity since weeks, I decieded to write here. I'm a bit lost with the examples. I just want to extract some depth infomations. But the only thing I can see is a black&red image where the red color is the same in very different depths. Best case would be to get an grayscale image, where the pixel show a mapped image from for example 1-5 meter, adjustable range if possible.

I bought the astra 2 with the hope to use it in an installation and buy more of them, but currently it's not useable for me.

I use openCV to extract infos from the image. Currently it's a webcam image, but it'a difficult environment and a depth sensor would work much better. I hope, someone can help.

noemis84, 2024-05-15:
I also don't understand, why I can choose under Profiles in the DepthProfile_800 the Format, but this is not working as I would expect. After switching from "PB_FORMAT_UNKNOWN" to "OB_FORMAT_GRAY" which seem to be my target, it just throws some errors.

zhonghong322, 2024-05-17:
Thank you for your feedback. I'll forward the issue to our Unity development engineer.

whichow, 2024-05-17:
> I also don't understand, why I can choose under Profiles in the DepthProfile_800 the Format, but this is not working as I would expect. After switching from "PB_FORMAT_UNKNOWN" to "OB_FORMAT_GRAY" which seem to be my target, it just throws some errors.

DepthProfile not support OB_FORMAT_GRAY format, OB_FORMAT_UNKNOWN means any format that support by DepthProfile, use this by default.

whichow, 2024-05-17:
> hey there, I wrote an issue in the unity section [here](https://github.com/orbbec/OrbbecUnitySDK/issues/3)
>
> Because there is no activity since weeks, I decieded to write here. I'm a bit lost with [...]

You can get the depth value from OrbbecFrameSource.GetDepthFrame().data, it's a depth value array, you can use it as depth image data with width and height, and normalize the value to 0~1 use OpenCV or any other method you like.

noemis84, 2024-05-17:
Thank you for the response. I'm not so good in coding to understand how to handle all these bytes... I'm more from the visible side of life... I think in pixel and am not so good on in my eyes complex coding stuff. But still, I manages to ask ChatGPT about it and now have a working example using also the recommended line.

Still I wonder, if body tracking is an option for the unity sdk like in [this demo](https://www.youtube.com/watch?v=olRi-k7sybM&ab_channel=Orbbec) from three years ago...

noemis84, 2024-05-17:
Meanwhile I have some ugly problems getting the app runing on windows... but this is another story and I'll open a new issue for this.

whichow, 2024-05-20:
> Thank you for the response. I'm not so good in coding to understand how to handle all these bytes... I'm more from the visible side of life... I think in pixel and am not so good on in my eyes [...]

I convert depth to Unity Texture like this:
```csharp
Texture2D depthTexture = new Texture2D(frame.width, frame.height, TextureFormat.RGB24, false);
ColorizeDepth(frame.data);
depthTexture.LoadRawTextureData(depthBuffer);
depthTexture.Apply();

private byte[] ColorizeDepth(byte[] depthData)
{
    int bufferSize = depthData.Length / 2 * 3;
    
    byte[] depthBuffer = new byte[bufferSize];
    
    int length = depthData.Length;
    for (int i = 0; i < length / 2; i++)
    {
        short depth = (short)((depthData[i * 2 + 1] << 8) + depthData[i * 2]);
        byte depthByte = (byte)0;
        if (depth != 0)
        {
            depthByte = (byte)(255 - (255 * depth / 10000.0f));
        }
        depthBuffer[i * 3 + 0] = depthByte;
        depthBuffer[i * 3 + 1] = depthByte;
        depthBuffer[i * 3 + 2] = depthByte;
    }

    return depthBuffer
}
```