orbbec/OrbbecSDK #88: Gemini 2XL [35472][ObRTPSink.cpp:320] Drop output-frame to receive new frame due to reclaimed-frame queue is empty

LLY13, 2024-05-16:
// Create a Context
    ob::Context ctx;

    // Enter the device ip address (currently only FemtoMega devices support network connection, and its default ip address is 192.168.1.10)
    std::string ip = "192.168.1.10";
    if (ip.empty()) {
        ip = "192.168.1.30";
    }

    // Create a network device through ip (the default port number is: 8090, devices that currently support network mode do not support modifying the port
    // number)
    auto device = ctx.createNetDevice(ip.c_str(), 8090);

    // pass in device to create pipeline
    auto pipe = std::make_shared<ob::Pipeline>(device);

    // Create Config for configuring Pipeline work
    std::shared_ptr<ob::Config> config = std::make_shared<ob::Config>();

    // Get the depth camera configuration list
    auto depthProfileList = pipe->getStreamProfileList(OB_SENSOR_DEPTH);
    // use default configuration
    auto depthProfile = depthProfileList->getVideoStreamProfile(640, 400, OB_FORMAT_Y16, 5);
    // enable depth stream
    config->enableStream(depthProfile);

    // Get the color camera configuration list
    auto colorProfileList = pipe->getStreamProfileList(OB_SENSOR_COLOR);
    // use default configuration
    auto colorProfile = colorProfileList->getVideoStreamProfile(1280, 800, OB_FORMAT_MJPG, 10);
    // enable depth stream
    config->enableStream(colorProfile);

    config->setAlignMode(ALIGN_D2C_HW_MODE);

    ob::PointCloudFilter pointCloud;
    auto cameraParam = pipe->getCameraParam();
    pointCloud.setCameraParam(cameraParam);
    //将 Depth 单位设置到点云 Filter
    pointCloud.setPositionDataScaled(1);
    //设置点云的类型，设置成输出 RGBD 点云
    pointCloud.setCreatePointFormat(OB_FORMAT_RGB_POINT);

    // Pass in the configuration and start the pipeline
    pipe->start(config);
    // Create a window for rendering and set the resolution of the window
    Window app("MultiDeviceViewer", 1280, 720, RENDER_ONE_ROW);

    while (app) {
        auto frameSet = pipe->waitForFrames(100);
        if (frameSet) {
            auto colorFrame = frameSet->getFrame(OB_FRAME_COLOR);
            auto depthFrame = frameSet->getFrame(OB_FRAME_DEPTH);
            if (colorFrame && depthFrame) {
                if (colorFrame->format() == OB_FORMAT_H264 || colorFrame->format() == OB_FORMAT_H265) {
                    // For H264 and H265 format image frames, users can refer to FFmpeg and other decoding libraries and their routines to complete decoding and
                    // rendering display. This example is to keep the project and code concise, so it will not be shown.
                    if (colorFrame->index() % 30 == 0) {
                        // Print the Color data frame information every 30 frames
                        std::cout << "Color Frame: index=" << colorFrame->index() << ", timestamp=" << colorFrame->timeStamp();
                    }
                    app.addToRender(depthFrame);
                }
                else {
                    //调用点云 Filter 生成点云数据
                    //std::shared_ptr<ob::Frame> frame = pointCloud.process(frameSet);
                    app.addToRender({ colorFrame, depthFrame });
                }
            }
        }

        // Sleep for 30ms, control the rendering display refresh rate
        std::this_thread::sleep_for(std::chrono::milliseconds(30));
        return ResultUtils::success();
    }

跑这个demo的时候老出现这个报错是什么原因

LLY13, 2024-05-16:
单网口连接

zhonghong322, 2024-05-21:
This error message indicates: The buffer in the RTSP queue is empty. Other parts are holding onto these frames and not releasing them, so memory is being forcibly freed. Are you processing point clouds in your sample? Do you encounter any issues when running the NetDevice sample?