orbbec/OrbbecSDK #89: Windows 10/11 problems using OrbbecUnitySDK after building

noemis84, 2024-05-17:
As the title says, I have problems after bulding the executable. 

My environment:
I currently use Unity 2022.3.25 and in editor everything works fine so far. I use an Astra 2.
I installed the newest plugin from git, which says 1.1.9 and the logs say that it's Orbbec SDK version: 1.8.1

I work on a win 10 machine.
The mini pc, where the app should run, comes with win 11. On the win 11 machine I installed the orbbec driver.

Building from Windows 10 with Mono runs fine on same Machine, my workhorse.

Same build on Windows 11 gives:

`Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/OrbbecSDK
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/OrbbecSDK.dll
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/OrbbecSDK.dll
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/OrbbecSDK
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/libOrbbecSDK
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/libOrbbecSDK.dll
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/libOrbbecSDK.dll
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/libOrbbecSDK
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/OrbbecSDK
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/OrbbecSDK.dll
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/OrbbecSDK.dll
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/OrbbecSDK
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/libOrbbecSDK
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/libOrbbecSDK.dll
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/libOrbbecSDK.dll
Fallback handler could not load library C:/Users/blendFX/Desktop/Haftgründe/Haftgründe_Data/MonoBleedingEdge/libOrbbecSDK
DllNotFoundException: OrbbecSDK assembly:<unknown assembly> type:<unknown type> member:(null)
  at (wrapper managed-to-native) Orbbec.obNative.ob_get_major_version()
  at Orbbec.Version.GetMajorVersion () [0x00000] in <db9bcc40717b4019b403ec5264d3634d>:0 
  at OrbbecUnity.OrbbecContext.InitSDK () [0x00000] in <9a668bd3f1224845a8565eb9ba0fee72>:0 
  at OrbbecUnity.OrbbecContext.Awake () [0x00008] in <9a668bd3f1224845a8565eb9ba0fee72>:0 
UnityEngine.GameObject:Internal_AddComponentWithType(GameObject, Type)
UnityEngine.GameObject:AddComponent(Type)
UnityEngine.GameObject:AddComponent()
OrbbecUnity.OrbbecContext:get_Instance()
OrbbecUnity.OrbbecDevice:Start()`

The .dll is indeed not there. It’s under “Haftgründe_Data\Plugins” - I wonder why the same build runs on Win10.
I run the build from an usb stick. So it’s the same executable.

On the Win 11 machine your Unity SDK Demo for Win also does not run.

So I switched from Mono to IL2CPP and now it also does not run on Win10. Getting this in the log:

``
NotSupportedException: IL2CPP does not support marshaling delegates that point to instance methods to native code. The method we're attempting to marshal is: Orbbec.Pipeline::OnFrameset
  at Orbbec.Pipeline.Start (Orbbec.Config config, Orbbec.FramesetCallback callback) [0x00000] in <00000000000000000000000000000000>:0 
  at UnityEngine.Events.UnityEvent.Invoke () [0x00000] in <00000000000000000000000000000000>:0 
  at OrbbecUnity.OrbbecPipeline.InitPipeline (Orbbec.Device device) [0x00000] in <00000000000000000000000000000000>:0 
  at UnityEngine.Events.InvokableCall`1[T1].Invoke (T1 args0) [0x00000] in <00000000000000000000000000000000>:0 
  at UnityEngine.Events.UnityEvent`1[T0].Invoke (T0 arg0) [0x00000] in <00000000000000000000000000000000>:0 
  at OrbbecUnity.OrbbecDevice+<WaitForDevice>d__8.MoveNext () [0x00000] in <00000000000000000000000000000000>:0 
  at UnityEngine.SetupCoroutine.InvokeMoveNext (System.Collections.IEnumerator enumerator, System.IntPtr returnValueAddress) [0x00000] in <00000000000000000000000000000000>:0 
``
So I asume, there are some serious problems after building. I hope you can help, because my time is running and the installation at the end will use 6 of the Astra 2 and it should be solid.

The next days I won't be at the office, due weekend and holiday on monday, but I can provide more infos next week if needed.

obwh, 2024-05-20:
> As the title says, I have problems after bulding the executable.
>
> My environment: I currently use Unity 2022.3.25 and in editor everything works fine so far. I use an Astra 2. I installed the newest [...]

I checked Astra 2 on Win11 use this version SDK, it's work fine, can you provide your project for us to debug

noemis84, 2024-05-21:
I'll try this as soon, as I'm in the office. Hopefully tomorrow. To debug the project, should I invite you to the project on github?

And also important for the context: buildling is not the problem. Also both apps are starting. But I get no image on certain app/winversion combinations and these errors are coppied from the player log, not editor log.

obwh, 2024-05-21:
Our Unity SDK not support IL2CPP right now, so use Mono as script backend. You can upload the smallest example project includes our SDK that can reproduces the problem here, which is usually not too big.

noemis84, 2024-05-22:
Thank you for clearing this up. To better debug this I installed the latest patch release from Unity 2022.3.29 and created an emtpy project, imported you latest unity sdk, added the depth sample scene to the build settings. Here are the results:

Win 10 machine runs the application without problems.
Win 11 mini PC runs the application, but no image from orbbec... AND
Win 11 notebook (new in this testsetup) runs the application and shows the image.

Bot Win 11 machines say, that there are no windows updates. 23H2.

At this point it's getting strange. So I can confirm, as you, that win 11 itself can't be the problem. I also think, that sharing my project from this point makes no sense, because it's your own depth sample scene with the newest unity LTS version. Nothing special.

I called my testapp "OrbbecAstra2Win11Bug" and I attach the both player logs.

[Player_miniPC.log](https://github.com/orbbec/OrbbecSDK/files/15399484/Player_miniPC.log)
[Player_notebook.log](https://github.com/orbbec/OrbbecSDK/files/15399486/Player_notebook.log)

as you can see, even if's the same app starting from a usb stick (D:) on different win 11 machines, it acts different. I think the main problem seem to be in the path:

I searches here:
D:/OrbbecAstra2Win11Bug/OrbbecAstra2Win11Bug_Data/MonoBleedingEdge/OrbbecSDK

But manual searching shows me the dll here:
D:\OrbbecAstra2Win11Bug\OrbbecAstra2Win11Bug_Data\Plugins\x86_64

Some lines above there is another *.dll, which could not be loaded:
XInput1_3.dll not found. Trying XInput9_1_0.dll instead...

I think this causes not the orbbec bug, but could be the same root...

I hope, you've an idea. Maybe some sort of net library dependencie?

noemis84, 2024-05-22:
I asked again ChatGPT about this and it recommended to install the latest Microsoft Visual C++ Redistributable Version from [here](https://learn.microsoft.com/en-US/cpp/windows/latest-supported-vc-redist?view=msvc-170) and now it seem to work. Maybe it's a good idea to mention this somewhere in the readme or also in the log... that'd save me many hours.

What do you think? Or did I overread something? Especially a hint on the home page from the OrbbecUnitySDK on git would be helpful, because the workstations mostly come with many packages due there software. But in such case with a fresh mini pc the hint would be very helpful.