orbbec/OrbbecSDK #90: Firmware update failure on macOS, device is unresponsive

philnelson, 2024-05-18:
I attempted to update the firmware on a Gemini 2 and it appaared to work, however, upon reboot it failed and now I cannot access any streams or get it to enumerate.

Last log lines:

```
Upgrade firmware complete
Reboot device
[05/17 19:16:00.360043][error][3958399][ObLibuvcDevicePort.cpp:282] getXu failed, error code=-1
[05/17 19:16:00.360351][error][3958399][HostProtocol.cpp:420] Set property value failed! propertyId:29, rc:{statusCode: 2, respErrorCode: 65535, msg: receive control transfer failed!}
[05/17 19:16:00.360412][warning][3958399][ObException.hpp:40] Request failed, statusCode: 2, msg: receive control transfer failed!
[05/17 19:16:00.368388][error][3958405][ObLibuvcDevicePort.cpp:257] setXu failed, error code=-4
[05/17 19:16:00.368417][info][3958399][Gemini2Device.cpp:160] Gemini2 device destroyed! PID:1648, SN:AY3JB20004B, depthMode: {name: Unbinned Dense Default, checksum(hex): ee24bacde1a916f894b2aceec941b500, optionCode: NORMAL}
[05/17 19:16:00.368472][error][3958405][HostProtocol.cpp:562] get firmware data failed. propertyId: 1037, rc: {statusCode: 2, respErrorCode: 65535, msg: send control transfer failed!}
[05/17 19:16:00.368636][warning][3958405][ObException.hpp:40] Request failed, statusCode: 2, msg: send control transfer failed!
[05/17 19:16:00.368904][info][3958624][DeviceManager.cpp:181] Device changed! removed: 1, added: 0
[05/17 19:16:00.368932][info][3958624][DeviceManager.cpp:15] Removed device(s) list: (1)
[05/17 19:16:00.368943][info][3958624][DeviceManager.cpp:24] 	- Name: Orbbec Gemini 2, PID: 0x0670, SN/ID: 01234567890, Connection: USB2.0
[05/17 19:16:00.368956][info][3958624][DeviceManager.cpp:15] Current device(s) list: (0)
[05/17 19:16:00.424715][info][3958399][ObLibuvcDevicePort.cpp:70] uvc_close done.
[05/17 19:16:00.424743][info][3958399][ObLibuvcDevicePort.cpp:71] ~ObLibuvcDevicePort done
[05/17 19:16:03.432480][info][3958399][Context.cpp:87] Context destroyed
function:reboot
args:nullptr
message:Request failed, statusCode: 2, msg: receive control transfer failed!
type:6
```

I have my Mac configured to "Ask" before mounting USB devices, and I think that this functionality may have thrown a wrench in the Orbbec update process, basically bricking he device (as of right now).

Can anyone suggest next steps or other diagnostic?

zhonghong322, 2024-05-21:
To check the status of the device on MacOS, please run the following command:  ioreg -p IOUSB -w0 -l

zhonghong322, 2024-05-21:
please run the following command: ioreg -p IOUSB -w0 -l
[image]

zhonghong322, 2024-05-21:
@philnelson  We have not been able to reproduce the issue you reported. Have you performed cross-testing?

Can the device be recognized correctly when hot-plugged?
Does the device also exhibit issues on Windows or Linux?