orbbec/OrbbecSDK #91: Help, Question aboult timestamp

hplegend, 2024-05-26:
what's the different between  "get_timestamp" and "get_system_timestamp"? I print the two timestamp, and found get_system_timestamp ahead of get_timestamp.  On the other hand, get_timestamp need manual config 'aync' at orbbec viewer or it will return 0.

natelowry, 2024-05-28:
i believe `get_timestamp` returns the timestamp from the camera and  `get_system_timestamp` is from the host computer.  on many cameras the clock on the camera does not persist if it is unplugged hence having to sync it after startup.  i want to say some models of cameras start a timer from startup time too, but don't remember which ones did that.

zhonghong322, 2024-06-04:
> i believe `get_timestamp` returns the timestamp from the camera and `get_system_timestamp` is from the host computer. on many cameras the clock on the camera does not persist if it is unplugged [...]

Yes, what you said is correct. The get_timestamp returns the timestamp from the camera, while the get_system_timestamp returns the timestamp from the host computer. The timestamps from the camera and the host computer might not be consistent. You can synchronize them by using the time synch interface to ensure that the camera's timestamp matches the host computer's timestamp.

closed 2024-09-19 by zhonghong322