orbbec/OrbbecSDK #94: Are transformation functions thread safe?

tory0807, 2024-06-05:
I was trying to use 'transformationDepthFrameToColorCamera()' to get multiple depth maps at different resolutions. Because transformation of depth maps takes time, I had to use multiple threads to call it . But in this case, the application crashed.

Is it possible to call 'transformationDepthFrameToColorCamera()' from multiple threads at the same time?
Please check the test code:
```#include <libobsensor/ObSensor.hpp>
#include <libobsensor/hpp/Utils.hpp>
#include <windows.h>
#include <fstream>
#include <iostream>
#include <mutex>
#include <string>
#include <queue>

std::shared_ptr<ob::Context> ctx;
std::shared_ptr<ob::Device>   device;
std::shared_ptr<ob::Pipeline> pipeline;
std::vector<std::thread> workerThreadList;
std::recursive_mutex          deviceMutex;
std::mutex frameMutex;
std::condition_variable waitFrame;
std::queue<std::shared_ptr<ob::FrameSet>> queuedFrames;

bool streamStarted = false;
bool working = true;
int frameCount = 0;

void handleDeviceConnected(std::shared_ptr<ob::DeviceList> connectList);
void handleDeviceDisconnected(std::shared_ptr<ob::DeviceList> disconnectList);
void handleDeviceConnected(std::shared_ptr<ob::DeviceList> connectList);
void handleDeviceDisconnected(std::shared_ptr<ob::DeviceList> disconnectList);
void enqueueFrameset(std::shared_ptr<ob::FrameSet> frameset);
void handleFrameset(std::shared_ptr<ob::FrameSet> frameset);
void startStream();
void inputWatcher();
void quitWorkers();
void handleFrame();

int main() try
{
    ob::Context::setLoggerSeverity(OB_LOG_SEVERITY_ERROR);
    ctx = std::make_shared<ob::Context>();

    // Register device callback
    ctx->setDeviceChangedCallback([](std::shared_ptr<ob::DeviceList> removedList, std::shared_ptr<ob::DeviceList> addedList) {
        handleDeviceDisconnected(removedList);
        handleDeviceConnected(addedList);
        });

    // Query the list of connected devices.
    std::shared_ptr<ob::DeviceList> devices = ctx->queryDeviceList();

    // Handle connected devices（and open one device）
    handleDeviceConnected(devices);

    if (!device) {
        std::cout << "Waiting for connect device..." << std::endl;
        while (!device) {
            std::this_thread::sleep_for(std::chrono::milliseconds(100));
        }
    }

    std::cout << "Input 'quit' or 'q' to stop application." << std::endl;
    std::thread inputThread(inputWatcher);

    while (working) {
        std::this_thread::sleep_for(std::chrono::milliseconds(1000));
        std::cout << "frames handled: " << frameCount << std::endl;
    }

    inputThread.join();
[... 223 lines cut ...]
{
    waitFrame.notify_all();
    for (auto& t : workerThreadList)
    {
        t.join();
    }
    workerThreadList.clear();

    while (!queuedFrames.empty()) {
        queuedFrames.pop();
    }
}

void inputWatcher() {
    while (true) {
        std::string cmd;
        std::cout << "\nInput command:  ";
        std::getline(std::cin, cmd);
        if (cmd == "quit" || cmd == "q") {
            std::cout << "\nQuit the application. Please wait a moment ..." << std::endl << std::endl;
            break;
        }
    }
    working = false;
}```

cool-wuzh, 2024-06-06:
Hi tory:
    I tried to run the code and the crash log is 'Alloc frame buffer failed!', the reason for this is caused by caching frames all the time in the code, because the size of the memory pool inside the Orbbec SDK defaults to 2GB, so exceeding the threshold causes the request failure issue. It is suggested to put a limit on the number of queues.
``` C++
void enqueueFrameset(std::shared_ptr<ob::FrameSet> frameset) {
    std::unique_lock<std::mutex> lock(frameMutex);
    if(queuedFrames.size() > 10){
        queuedFrames.pop();
    }
    queuedFrames.push(std::move(frameset));
    waitFrame.notify_one();
}
```

tory0807, 2024-06-06:
I followed your recommendations and especially copied what you wrote, but the application still crashes.

cool-wuzh, 2024-06-06:
I tried to run the modified program for a while without crash, can you provide the crash stack or detailed log information?
[image]

tory0807, 2024-06-07:
This is the log from windows event viewer.
```xml
<?xml version="1.0" encoding="UTF-16"?>
<WERReportMetadata>
	<OSVersionInformation>
		<WindowsNTVersion>10.0</WindowsNTVersion>
		<Build>22000</Build>
		<Product>(0x30): Windows 10 Pro</Product>
		<Edition>Professional</Edition>
		<BuildString>22000.2245.amd64fre.co_release.210604-1628</BuildString>
		<Revision>2245</Revision>
		<Flavor>Multiprocessor Free</Flavor>
		<Architecture>X64</Architecture>
		<LCID>1042</LCID>
		<BuildLayers>
			<BuildLayer LayerName="DesktopEditions" OsMajorVersion="10" OsMinorVersion="0" BuildNumber="22000" QfeNumber="2245" IsTopLevel="1" IsChecked="0" BuildStamp="210604-1628" BuildBranch="co_release" BuildArch="amd64" BuildLab="22000.co_release.210604-1628" BuildLabEx="22000.1.amd64fre.co_release.210604-1628"/>
			<BuildLayer LayerName="OnecoreUAP" OsMajorVersion="10" OsMinorVersion="0" BuildNumber="22000" QfeNumber="1" IsTopLevel="0" IsChecked="0" BuildStamp="210604-1628" BuildBranch="co_release" BuildArch="amd64" BuildLab="22000.co_release.210604-1628" BuildLabEx="22000.1.amd64fre.co_release.210604-1628"/>
			<BuildLayer LayerName="ShellCommon" OsMajorVersion="10" OsMinorVersion="0" BuildNumber="22000" QfeNumber="2245" IsTopLevel="0" IsChecked="0" BuildStamp="230719-0901" BuildBranch="co_release_svc_prod3" BuildArch="amd64" BuildLab="22000.co_release_svc_prod3.230719-0901" BuildLabEx="22000.2245.amd64fre.co_release_svc_prod3.230719-0901"/>
		</BuildLayers>
	</OSVersionInformation>
	<ProblemSignatures>
		<EventType>APPCRASH</EventType>
		<Parameter0>OrbbecSample.exe</Parameter0>
		<Parameter1>0.0.0.0</Parameter1>
		<Parameter2>66625a18</Parameter2>
		<Parameter3>StackHash_69b3</Parameter3>
		<Parameter4>10.0.22000.2124</Parameter4>
		<Parameter5>f7fdb45f</Parameter5>
		<Parameter6>c0000374</Parameter6>
		<Parameter7>PCH_F0_FROM_ntdll+0x00000000000A4754</Parameter7>
	</ProblemSignatures>
	<DynamicSignatures>
		<Parameter1>10.0.22000.2.0.0.256.48</Parameter1>
		<Parameter2>1042</Parameter2>
		<Parameter22>69b3</Parameter22>
		<Parameter23>69b307ffd45205023a8786efa28c1303</Parameter23>
		<Parameter24>c034</Parameter24>
		<Parameter25>c0349e8eceefbeaf7dc06ef22dcc6c9c</Parameter25>
	</DynamicSignatures>
	<SystemInformation>
		<MID>1A829B6E-DBF1-4815-B09A-F00A660ECC90</MID>
		<MarkerFile>1462_MSI_Raider GE76 12UE_13.TCA4.J28.16.0.8</MarkerFile>
		<SystemManufacturer>Micro-Star International Co., Ltd.</SystemManufacturer>
		<SystemProductName>Raider GE76 12UHS</SystemProductName>
		<BIOSVersion>E17K4IMS.206</BIOSVersion>
		<OSInstallDate>1664610135</OSInstallDate>
		<OSInstallTime>1601-01-02T22:58:38Z</OSInstallTime>
		<TimeZoneBias>-09:00</TimeZoneBias>
	</SystemInformation>
	<DAMInformation>
		<LastEnterCSTime>133619317892075094</LastEnterCSTime>
		<LastLeaveCSTime>133619317892175156</LastLeaveCSTime>
		<LastS0SuspendTime>133613272385714239</LastS0SuspendTime>
		<LastS0ResumeTime>133613272965105787</LastS0ResumeTime>
	</DAMInformation>
	<SecureBootState>
		<UEFISecureBootEnabled>1</UEFISecureBootEnabled>
		<PolicyPublisher>{77fa9abd-0359-4d32-bd60-28f4e78f784b}</PolicyPublisher>
	</SecureBootState>
	<ReportInformation>
		<Guid>4ce7b0e9-452d-4001-aa42-047d3347f37f</Guid>
		<CreationTime>2024-06-07T00:55:17Z</CreationTime>
	</ReportInformation>
</WERReportMetadata>
```
Here is the last part of the log file.
```
[06/07 09:55:17.045647][debug][36444][FrameProcessingBlock.cpp:26] Block created with frame queue size: 10
[06/07 09:55:17.073864][debug][8928][PropertyAccessor.cpp:28] get property value success! propertyId: 14, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.074095][debug][8928][PropertyAccessor.cpp:28] get property value success! propertyId: 15, value: {intValue: 0, floatValue: 0}
[... 175 lines cut ...]
[06/07 09:55:17.445167][debug][23496][PropertyAccessor.cpp:28] get property value success! propertyId: 15, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445280][debug][23496][PropertyAccessor.cpp:28] get property value success! propertyId: 118, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445355][debug][23496][PropertyAccessor.cpp:28] get property value success! propertyId: 14, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445487][debug][33480][PropertyAccessor.cpp:28] get property value success! propertyId: 14, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445562][debug][33480][PropertyAccessor.cpp:28] get property value success! propertyId: 15, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445669][debug][33480][PropertyAccessor.cpp:28] get property value success! propertyId: 118, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445776][debug][33480][PropertyAccessor.cpp:28] get property value success! propertyId: 14, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445843][debug][33480][PropertyAccessor.cpp:28] get property value success! propertyId: 15, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445894][debug][33480][PropertyAccessor.cpp:28] get property value success! propertyId: 118, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445949][debug][33480][FrameProcessingBlock.cpp:26] Block created with frame queue size: 10
[06/07 09:55:17.445980][debug][23496][PropertyAccessor.cpp:28] get property value success! propertyId: 15, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.445986][debug][33480][FrameProcessingBlock.cpp:26] Block created with frame queue size: 10
[... cut ...]
[06/07 09:55:17.475545][debug][29948][PropertyAccessor.cpp:28] get property value success! propertyId: 15, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.475590][debug][29948][PropertyAccessor.cpp:28] get property value success! propertyId: 118, value: {intValue: 0, floatValue: 0}
[06/07 09:55:17.475634][debug][29948][FrameProcessingBlock.cpp:26] Block created with frame queue size: 10
[06/07 09:55:17.475669][debug][29948][FrameProcessingBlock.cpp:26] Block created with frame queue size: 10

```

tory0807, 2024-06-07:
Here is the call stack:

[image]

cool-wuzh, 2024-06-07:
Can you see which line of code crashed? Is it the frameset or device with a null pointer?

tory0807, 2024-06-07:
Please check this code with line numbers.

[image]

tory0807, 2024-06-07:
This is the call stack with symbols.

[image]

cool-wuzh, 2024-06-07:
I may know what the problem is, it should be a bug. please modified to call transformationDepthFrameToColorCamera only once and check if is problem.

tory0807, 2024-06-07:
I was able to run it a little longer, but it eventually crashed.

[image]

cool-wuzh, 2024-06-07:
I've reproduced the bug, and it is indeed due to a multi-threaded resource access conflict issue, which we'll fix in the next SDK release version.

zhonghong322, 2024-08-28:
> I was able to run it a little longer, but it eventually crashed. [...]

We have made modifications. Can you test using OrbbecSDK version 1.10.12 to see if the issue is resolved?

tory0807, 2024-08-28:
Yes, it looks fine. Thank you!!

closed 2024-08-28 by tory0807