orbbec/OrbbecSDK #96: Compatibility firmware Femto Bolt for Orbbec SDK > v1.10.2

ChuckM90, 2024-06-05:
From Orbbec SDK 1.10.2, Femto Bolt needs to upgrade to firmware 1.1.1. This firmware is no more avaliable in the repo "Orbbec firmware" from 28 May [https://github.com/orbbec/OrbbecFirmware/commit/b27b443d560e41e50e58c83e0af22caa945726f2](url)
Since this is causing a bit of confusion, which firmware for Femto Bolt is compatible for this SDK version and above?

zhonghong322, 2024-06-05:
Sorry ,There is a bug in Femto Bolt v1.1.1, so I have taken it down. We will release a new version this week.

GestureArtAndDesign, 2024-06-05:
> Sorry ,There is a bug in Femto Bolt v1.1.1, so I have taken it down. We will release a new version this week.

What is the nature of the bug?

zhonghong322, 2024-06-07:
> From Orbbec SDK 1.10.2, Femto Bolt needs to upgrade to firmware 1.1.1. This firmware is no more avaliable in the repo "Orbbec firmware" from 28 May [orbbec/OrbbecFirmware@b27b443](https://github.com/orbbec/OrbbecFirmware/commit/b27b443d560e41e50e58c83e0af22caa945726f2) Since this is causing a bit of confusion, which firmware for Femto Bolt is compatible for this SDK version and above?

New firmware v1.1.2 has released：
https://github.com/orbbec/OrbbecFirmware/releases/tag/Femto-Bolt-Firmware

zhonghong322, 2024-06-07:
> > Sorry ,There is a bug in Femto Bolt v1.1.1, so I have taken it down. We will release a new version this week.
> 
> What is the nature of the bug?

timestamp overflow