orbbec/OrbbecSDK #98: OrbbecViewer 无法识别USB 3.2 gen1 和 雷电4的转接USB3.0

zealX, 2024-06-13:
链接GEMINI 335采集数据显示 USB2.1，无法获取较大的分辨率原始数据

zealX, 2024-06-13:
[image]

zealX, 2024-06-13:
配置参数输出也是错的：选择上图这个配置可以看到深度实际给出的分辨率却是1920x1080。输出的配置内容如下：[ColorIntrinsic]
fx=1034.536011
fy=1034.449707
cx=969.158630
cy=535.346741
width=1920
height=1080
[ColorDistortion]
k1=0.007847
k2=-0.047242
k3=0.032096
k4=0.000000
k5=0.000000
k6=0.000000
p1=0.000362
p2=-0.000044

[DepthIntrinsic]
fx=1034.536011
fy=1034.449707
cx=969.158630
cy=535.346741
width=1920
height=1080
[DepthDistortion]
k1=0.000000
k2=0.000000
k3=0.000000
k4=0.000000
k5=0.000000
k6=0.000000
p1=0.000000
p2=0.000000

[D2CTransformParam]
rot0=1.000000
rot1=0.000000
rot2=0.000000
rot3=0.000000
rot4=1.000000
rot5=0.000000
rot6=0.000000
rot7=0.000000
rot8=1.000000
trans0=0.000000
trans1=0.000000
trans2=0.000000

zhonghong322, 2024-06-13:
> 配置参数输出也是错的：选择上图这个配置可以看到深度实际给出的分辨率却是1920x1080。输出的配置内容如下：[ColorIntrinsic] fx=1034.536011 fy=1034.449707 cx=969.158630 cy=535.346741 width=1920 height=1080 [ColorDistortion] k1=0.007847 k2=-0.047242 [...]

这个内参是 D2C后的内参，D2C后，Depth的分辨率会转换到 Color的分辨率。

zhonghong322, 2024-06-13:
> 链接GEMINI 335采集数据显示 USB2.1，无法获取较大的分辨率原始数据

OrbbecViewer 无法识别USB 3.2 gen1 和 雷电4的转接USB3.0，这个是什么意思。 是通过雷电的TypeC的线，然后接在一个hub，再转接成TypeA的 接Gemini 335， 还是雷电的TypeC的线直接 接Gemini 335。

zealX, 2024-06-14:
> > 链接GEMINI 335采集数据显示 USB2.1，无法获取较大的分辨率原始数据
> 
> OrbbecViewer 无法识别USB 3.2 gen1 和 雷电4的转接USB3.0，这个是什么意思。 是通过雷电的TypeC的线，然后接在一个hub，再转接成TypeA的 接Gemini 335， 还是雷电的TypeC的线直接 接Gemini 335。

PC环境：win11
电脑：thinkpad t14p gen2
USB口 USB-A 3.2 gen1
雷电口 雷电4
状况说明：使用最新的OrbbecViewer链接335后，尝试直接连接USB3.2口，结果软件上显示USB2.1；然后尝试使用雷电拓展坞上的USB3.0，发现依然识别为USB2.1，如前述图片！

谢谢

zhonghong322, 2024-06-14:
我们使用NUC PC和其他带有雷电接口的ThinkPad笔记本做了验证，均可以正常识别。
我们推测是C-C线材的问题，希望换一根别的C-C雷电线材使用。
这里有我们推荐的线材https://vcp.developer.orbbec.com.cn/documentation
另外。在使用USB-A口连接相机时，需要快速插入，不然会有概率识别成USB 2.1。如果仍无法解决，请联系奥比售后进行换货处理。

closed 2024-06-24 by zealX