orbbec/OrbbecSDK #99: Femto Mega RGB stream stops after few seconds on Raspbery Pi 4

Bercon, 2024-06-17:
Femto Mega seems to stop reading RGB camera after running for ~10 seconds on Raspbery Pi 4. The device was powered via the separate power source using barreljack and connected to RPi4 with USB3 A -> USB C cable

```
$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
$ ./OBQuickStart
[06/17 12:27:49.264810][info][174354][Context.cpp:68] Context created with config: default config!
[06/17 12:27:49.265428][info][174354][Context.cpp:73] Work directory=/home/spectacularai/git/OrbbecSDK/build/bin, SDK version=v1.10.8-20240604-d1671a3
[06/17 12:27:49.265902][info][174354][LinuxPal.cpp:32] createObPal: create LinuxPal!
[06/17 12:27:49.276705][info][174354][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[06/17 12:27:49.277353][info][174354][DeviceManager.cpp:15] Current found device(s): (1)
[06/17 12:27:49.277741][info][174354][DeviceManager.cpp:24] 	- Name: Femto Mega, PID: 0x0669, SN/ID: CL2AA310061, Connection: USB3.2
[06/17 12:27:49.278174][info][174354][Pipeline.cpp:15] Try to create pipeline with default device.
[06/17 12:27:49.888082][info][174354][AbstractDevice.cpp:117] 	- Firmware version: 1.2.8
[06/17 12:27:49.894934][warning][174354][ObException.hpp:40] Request failed, statusCode: 4, msg: 
[06/17 12:27:49.895407][warning][174354][FemtoMegaAlgParamManager.cpp:88] Execute failure! A libobsensor_exception has occurred!
	 - where:88#FemtoMegaAlgParamManager
	 - msg:Request failed, statusCode: 4, msg: 
	 - type:N11libobsensor12io_exceptionE
[06/17 12:27:49.895701][error][174354][FemtoMegaAlgParamManager.cpp:88] Get imu calibration params failed!
[06/17 12:27:49.895855][warning][174354][FemtoMegaAlgParamManager.cpp:94] Get imu calibration param failed!load default param.
[06/17 12:27:49.898024][warning][174354][ObException.hpp:40] Request failed, statusCode: 4, msg: 
[06/17 12:27:49.898336][warning][174354][IFemtoMegaDevice.cpp:75] Execute failure! A libobsensor_exception has occurred!
	 - where:75#initDepthProcessParam
	 - msg:Request failed, statusCode: 4, msg: 
	 - type:N11libobsensor12io_exceptionE
[06/17 12:27:49.898595][warning][174354][IFemtoMegaDevice.cpp:75] get imu calibration param failed!load default param.
[06/17 12:27:50.204974][info][174354][FemtoMegaUvcDevice.cpp:49] FemtoMega UVC device created! PID: 0x0669, SN: CL2AA310061
[06/17 12:27:50.205199][info][174354][DeviceManager.cpp:150] Device created successfully! Name: Femto Mega, PID: 0x0669, SN/ID: CL2AA310061
[06/17 12:27:50.205385][info][174354][Pipeline.cpp:47] Pipeline created with device: {name: Femto Mega, sn: CL2AA310061}, @0x5580CFD410
[06/17 12:27:50.218843][info][174354][FemtoMegaUvcDevice.cpp:383] Color sensor has been created!
[06/17 12:27:50.219780][info][174354][VideoSensor.cpp:384] Query stream profile! size= 108, SensorType=OB_SENSOR_COLOR
[06/17 12:27:50.219946][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1920, height: 1080, fps: 15}
[06/17 12:27:50.220084][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 30}
[06/17 12:27:50.220215][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 25}
[06/17 12:27:50.220243][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 15}
[06/17 12:27:50.220472][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 5}
[06/17 12:27:50.220632][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 960, fps: 30}
[06/17 12:27:50.220761][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 960, fps: 25}
[06/17 12:27:50.220789][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 960, fps: 15}
[06/17 12:27:50.221024][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 960, fps: 5}
[06/17 12:27:50.221052][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1920, height: 1080, fps: 30}
[06/17 12:27:50.221280][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1920, height: 1080, fps: 25}
[06/17 12:27:50.221308][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1920, height: 1080, fps: 15}
[06/17 12:27:50.221538][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1920, height: 1080, fps: 5}
[06/17 12:27:50.221566][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 2560, height: 1440, fps: 25}
[06/17 12:27:50.221797][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 2560, height: 1440, fps: 15}
[06/17 12:27:50.221825][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 2560, height: 1440, fps: 5}
[06/17 12:27:50.222058][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 3840, height: 2160, fps: 25}
[06/17 12:27:50.222187][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 3840, height: 2160, fps: 15}
[06/17 12:27:50.222302][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 3840, height: 2160, fps: 5}
[06/17 12:27:50.222417][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 30}
[06/17 12:27:50.222528][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 25}
[06/17 12:27:50.222636][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 15}
[06/17 12:27:50.222746][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 5}
[06/17 12:27:50.222856][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 30}
[06/17 12:27:50.222964][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 25}
[06/17 12:27:50.223074][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 15}
[06/17 12:27:50.223181][info][174354][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 5}
[... 101 lines cut ...]
[06/17 12:27:50.582002][info][174354][Pipeline.cpp:277] Pipeline start done!
Color Frame: 1920x1080 8253831000 us
Depth Frame: 640x576 8253897999 us
Color Frame: 1920x1080 8254831000 us
Depth Frame: 640x576 8254897999 us
Color Frame: 1920x1080 8255897000 us
Depth Frame: 640x576 8255964000 us
Color Frame: 1920x1080 8256963000 us
Depth Frame: 640x576 8257030000 us
Color Frame: 1920x1080 8257963000 us
Depth Frame: 640x576 8258030000 us
Color Frame: 1920x1080 8258962000 us
Depth Frame: 640x576 8259029000 us
Depth Frame: 640x576 8259962000 us
Depth Frame: 640x576 8260962000 us
Depth Frame: 640x576 8262028000 us
Depth Frame: 640x576 8263093999 us
Depth Frame: 640x576 8264160000 us
Depth Frame: 640x576 8265227000 us
Depth Frame: 640x576 8266226000 us
Depth Frame: 640x576 8267293000 us
Depth Frame: 640x576 8268358000 us
Depth Frame: 640x576 8269424000 us

```

zhonghong322, 2024-06-18:
Sorry ,We have not been able to reproduce the issues you mentioned on both Raspberry Pi 4 and Raspberry Pi 5. Below is a screenshot of Raspberry Pi 4.
[image]

What was the status of the Mega when you encountered the issue? Was the LED indicator functioning properly? Additionally, it is recommended to try using a different USB 3.0 cable for cross-verification.

zhonghong322, 2024-06-18:
Did this issue occur sporadically? Could you please provide the issue log

Bercon, 2024-06-18:
Thank for your quick response!

This seems to have been caused by systemd/user manager constantly restarting lightdm user session every 10 seconds. This starts/stops a lot of services and seems to interfere with Orbbec RGB camera stream. Possibly related to v4l2, I saw this on the system logs:
```Jun 18 12:08:42 berry pipewire[13509]: [0:10:06.086660015] [13564] ERROR UVC uvcvideo.cpp:463 Camera /base/scb/pcie@7d500000/pci@0,0/usb@0,0-2:1.2-2bc5:0669 (Orbbec Femto Mega 3D Camera: Or) doesn't expose any supported format```

After I stopped lightdm with `sudo systemctl stop lightdm` the problem went away

closed 2024-06-18 by Bercon