orbbec/OrbbecSDK #104: gemini2 get right or right IR error

AAA-Fei, 2024-07-04:
No required type sensor found! Normal depth work mode, sensorType: OB_SENSOR_IR_RIGHT

zhonghong322, 2024-07-08:
Gemini2 does not support acquiring left IR and right IR,  it can only support  IR stream (OB_SENSOR_IR). You can switch between the left infrared (IR) and right infrared (IR).
[image]

AAA-Fei, 2024-07-08:
> Gemini2 does not support acquiring left IR and right IR, it can only support IR stream (OB_SENSOR_IR). You can switch between the left infrared (IR) and right infrared (IR). [...]

thank you

closed 2024-07-08 by AAA-Fei