orbbec/OrbbecSDK #110: Release 1.10.11 contains invalid CMake instructions

bytebunny, 2024-07-26:
The x64 Linux release [1.10.11](https://github.com/orbbec/OrbbecSDK/releases/tag/v1.10.11) contains a README that has invalid CMake instructions:

```cmake
...
# find Orbbec SDK
set(OrbbecSDK_DIR "/your/path/to/OrbbecSDK")
find_package(OrbbecSDK REQUIRED)
```

The release does not contain a single `.cmake` file, so `find_package` call fails. Am I missing something?

The README has also a section on building the project, but there is no source code.

someilay, 2025-01-20:
+

troeggla, 2025-02-19:
I have the same issue. Also packages downloaded via `apt` don't have any `.cmake` files, making including the SDK in your own project not very portable.

Does anyone have any suggestions?