orbbec/OrbbecSDK #111: Femto Bolt, Camera Setting

Shahriarch, 2024-08-07:
I am using the example of PointCloud and made a build to capture ply 3D scans, how can I change the camera settings, resolution, field of view, depth resolution in the code?

zhonghong322, 2024-08-07:
There are two methods: 
The first one is to modify the .xml configuration file, 
[image]

and the other one is to set the resolution through code.
https://orbbec.github.io/OrbbecSDK/doc/tutorial/English/[OrbbecSDK_C++_API_user_guide-v1.0.pdf](https://orbbec.github.io/OrbbecSDK/doc/tutorial/English/OrbbecSDK_C++_API_user_guide-v1.0.pdf)

zhonghong322, 2024-09-19:
@Shahriarch Have your issues been resolved?

closed 2024-10-15 by Shahriarch