orbbec/OrbbecSDK #116: Found a usb port without interface url

AsWeSee, 2024-08-16:
Astra Pro Plus is connected and driver is installed. 
When I query list of devices 
`auto devList = ctx.queryDeviceList();`
It returns 0 devices found

There are these messages in log

```
[08/16 15:17:40.319651][debug][21908][DeviceManager.cpp:30] DeviceManager init ...
[08/16 15:17:40.319662][info][21908][MfPal.cpp:105] createObPal: create WinPal!
[08/16 15:17:40.319717][debug][21908][MfPal.cpp:110] WmfPal init ...
[08/16 15:17:40.357632][debug][21908][MfPal.cpp:117] WmfPal created!
[08/16 15:17:40.357661][debug][21908][DeviceManager.cpp:34] Enable USB Device Enumerator ...
[08/16 15:17:40.407487][debug][21908][EnumeratorLibusb.cpp:324] queryDevicesInfo done!
[08/16 15:17:40.407564][warning][21908][MfPal.cpp:262] Found a usb port without interface url! drop it!(device url=\\?\USB#VID_2BC5&PID_060F#7&196F16E9&0&1#{A5DCBF10-6530-11D2-901F-00C04FB951ED} interface index=0)
[08/16 15:17:40.407721][debug][21908][UsbDeviceEnumerator.cpp:163] Current usb device port list:
[08/16 15:17:40.407737][debug][21908][UsbDeviceEnumerator.cpp:166]  - \\?\USB#VID_2BC5&PID_050F&MI_00#8&2913CC43&0&0000#{E5323777-F976-4F5B-9B55-B94699C46E44}\GLOBAL | USB Camera
[08/16 15:17:40.407952][debug][21908][MfPal.cpp:216] Create WinEventDeviceWatcher!
[08/16 15:17:40.408092][debug][21908][UsbDeviceEnumerator.cpp:78] No matched usb device found!
[08/16 15:17:40.408115][info][21908][DeviceManager.cpp:15] Current found device(s): (0)
[08/16 15:17:40.408173][debug][21908][DeviceManager.cpp:52] DeviceManager construct done!
```

Windows 10 23H2 version 22631.4037 
Orbbek SDK version 1.10.12 

Question.  Is in necessary to install/configure something?

Context. 
Globally, Im trying to connect Astra Pro Plus to Unity. UnityWrapper had same error in logs so I tried to build C++ to figure it out.

Also1. I launched UnityWrapper with OrbbecSDK v1.10.8 on Astra linux. Works partially. IR and Depth revieced well, but not color. Here screenshot of device data provided by wrapper. Linux is not a goal, but at least camera is physically working.
[image]

Also2. Demo from OpenNI sdk with driver installed is working fine on Windows. But this sdk seems lacking support of color camera, and unity wrappers for OpenNI became dead 10 years ago so I cant test it in Unity

Also3. I installed metadata. 3 devices were found but " of them are Orbbec devices", that look not correct.
One key was installed. That did not affect search problem from title

[image]

JKComing, 2024-08-21:
Hello, if you use an Orbbec camera (especially the Astra series) and your Windows computer cannot recognize it normally, you can refer to the attached file for settings.I hope this can help you.
[Astra series  camera setting.pdf](https://github.com/user-attachments/files/16689362/Astra.series.camera.setting.pdf)

zhonghong322, 2024-08-23:
> Astra Pro Plus is connected and driver is installed. When I query list of devices `auto devList = ctx.queryDeviceList();` It returns 0 devices found
> 
> There are these messages in log
> 
> ```
> [08/16 15:17:40.319651][debug][21908][DeviceManager.cpp:30] DeviceManager init ...
> [08/16 15:17:40.319662][info][21908][MfPal.cpp:105] createObPal: create WinPal!
> [08/16 15:17:40.319717][debug][21908][MfPal.cpp:110] WmfPal init ...
> [08/16 15:17:40.357632][debug][21908][MfPal.cpp:117] WmfPal created!
> [08/16 15:17:40.357661][debug][21908][DeviceManager.cpp:34] Enable USB Device Enumerator ...
> [08/16 15:17:40.407487][debug][21908][EnumeratorLibusb.cpp:324] queryDevicesInfo done!
> [08/16 15:17:40.407564][warning][21908][MfPal.cpp:262] Found a usb port without interface url! drop it!(device url=\\?\USB#VID_2BC5&PID_060F#7&196F16E9&0&1#{A5DCBF10-6530-11D2-901F-00C04FB951ED} interface index=0)
> [08/16 15:17:40.407721][debug][21908][UsbDeviceEnumerator.cpp:163] Current usb device port list:
> [08/16 15:17:40.407737][debug][21908][UsbDeviceEnumerator.cpp:166]  - \\?\USB#VID_2BC5&PID_050F&MI_00#8&2913CC43&0&0000#{E5323777-F976-4F5B-9B55-B94699C46E44}\GLOBAL | USB Camera
> [08/16 15:17:40.407952][debug][21908][MfPal.cpp:216] Create WinEventDeviceWatcher!
> [08/16 15:17:40.408092][debug][21908][UsbDeviceEnumerator.cpp:78] No matched usb device found!
> [08/16 15:17:40.408115][info][21908][DeviceManager.cpp:15] Current found device(s): (0)
> [08/16 15:17:40.408173][debug][21908][DeviceManager.cpp:52] DeviceManager construct done!
> ```
> 
> Windows 10 23H2 version 22631.4037 Orbbek SDK version 1.10.12
> 
> Question. Is in necessary to install/configure something?
> 
> Context. Globally, Im trying to connect Astra Pro Plus to Unity. UnityWrapper had same error in logs so I tried to build C++ to figure it out.
> 
> Also1. I launched UnityWrapper with OrbbecSDK v1.10.8 on Astra linux. Works partially. IR and Depth revieced well, but not color. Here screenshot of device data provided by wrapper. Linux is not a goal, but at least camera is physically working. ![Pasted image 20240606141935](https://private-user-images.githubusercontent.com/7767761/358550699-134f48a6-1150-41bc-bcbe-4f4c9e49f948.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ0MTU3MTYsIm5iZiI6MTcyNDQxNTQxNiwicGF0aCI6Ii83NzY3NzYxLzM1ODU1MDY5OS0xMzRmNDhhNi0xMTUwLTQxYmMtYmNiZS00ZjRjOWU0OWY5NDgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MjNUMTIxNjU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjM4NDBkNjdmYzUzMzVlNDM2NGVkMTc1MmM4YWQ0ZDJkNTRiOTc4OGNhZWRhODFmNGJkOTdkOTAwYThiMmYyYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.bZzg55MWt8LJ-BRo0_V7NYvMRoKQwcmJftAIsN_P5YM)
> 
> Also2. Demo from OpenNI sdk with driver installed is working fine on Windows. But this sdk seems lacking support of color camera, and unity wrappers for OpenNI became dead 10 years ago so I cant test it in Unity
> 
> Also3. I installed metadata. 3 devices were found but " of them are Orbbec devices", that look not correct. One key was installed. That did not affect search problem from title
> 
> ![Pasted image 20240816151433](https://private-user-images.githubusercontent.com/7767761/358550856-9b052113-16c9-4455-a2fd-b06e0120d35f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ0MTU3MTYsIm5iZiI6MTcyNDQxNTQxNiwicGF0aCI6Ii83NzY3NzYxLzM1ODU1MDg1Ni05YjA1MjExMy0xNmM5LTQ0NTUtYTJmZC1iMDZlMDEyMGQzNWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MjNUMTIxNjU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTFhM2YxOWFiMjE1ZTI4OTFlZTdmMzY3OTBmNzVhYjhhZGZlMWMwMDZkZWMyMWZhZjg5NWUzN2NjMmFlYmIxNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.VnZiFuIXAYWrmIpHpt_489tr9GObq7yMKbE3ixw_zx8)

Did you test using the OrbbecViewer tool? Does it work correctly?
https://github.com/orbbec/OrbbecSDK/releases/tag/v1.10.12

nataliadolina, 2024-11-25:
Hello! Did you solve the problem? I have the exact same issue

jujimeizuo, 2025-01-05:
I can run on ubuntu22.04, but failed on Win11. May be not compatible with Win11.