orbbec/OrbbecSDK #117: How to increase FrameBuffer size?

HunorLaczko, 2024-08-19:
I get the following error when I'm trying to use multiple Femto Bolt devices simultaneously:
`[FrameBufferManager.cpp:43] Alloc frame buffer failed! Out of max size! require=6.358MB, total usage: allocated=2046.477MB, max limit=2048.000MB`

I'm using a modified version of the MultiDeviceSync example where I save the outputs to the disk instead of streaming. It works with two Bolt devices, but with 4 I get this error. The original example with streaming works. I have increased the usbfs limit already. 

The OrbbecSDKConfig_v1.0.xml has an option to change the buffer size, but changing this value doesn't seem to change anything. 
`<MaxFrameBufferSize> 2048 </MaxFrameBufferSize>`

Could you help me on how I can increase this value?

zhonghong322, 2024-08-23:
Your method of modifying the buffer size is correct. After changing the buffer size, if you still encounter memory allocation issues, check the logs to see if the "max limit" reflects the value you modified. If the frame buffer cannot be allocated, it might be because your application is holding onto the frame buffer without releasing it.

HunorLaczko, 2024-08-23:
Thanks for your reply. My problem is that even though I changed the max limit, it doesn't reflect in the logs, there the max value says it's still the same. Similarly, I tried to turn off <GlobalTimestampFitterEnable>, but it didn't work. 

I downloaded the SDK from the release section for linux_x64, and I'm modifying this file: SDK/config/OrbbecSDKConfig_v1.0.xml

Do you have any idea why my application is not registering the changed values?

hzcyf, 2024-08-23:
> Thanks for your reply. My problem is that even though I changed the max limit, it doesn't reflect in the logs, there the max value says it's still the same. Similarly, I tried to turn off , but it didn't work.
> 
> I downloaded the SDK from the release section for linux_x64, and I'm modifying this file: SDK/config/OrbbecSDKConfig_v1.0.xml
> 
> Do you have any idea why my application is not registering the changed values?

`SDK/config/OrbbecSDKConfig_v1.0.xml`  is just a template file.  The config file should be placed into work directory (cwd) of you application.

zhonghong322, 2024-08-27:
> Thanks for your reply. My problem is that even though I changed the max limit, it doesn't reflect in the logs, there the max value says it's still the same. Similarly, I tried to turn off , but it [...]

How is it going now? Has the problem been solved?

HunorLaczko, 2024-08-27:
Thank you, this solved my problem!

closed 2024-08-27 by HunorLaczko