orbbec/OrbbecSDK #119: Failed to load depth engine plugin

jyhflash, 2024-09-12:
运行DepthViewer出现这个错误要怎么解决呢

zhonghong322, 2024-09-14:
你用的什么操作系统，Opengl的版本是什么，显卡类型是什么，有独立显卡吗？

xiaobai123-cell, 2024-09-18:
您好，我也出现了这个问题，一直没有解决。我用的是windows系统，Opengl的版本是3.1.7（python环境是3.9.13），显卡类型是NVIDIA GeForce RTX 4050 Laptop GPU，有独立显卡
[image]

zhonghong322, 2024-09-18:
Have you copied the depthengine_2_0.dll library file to the sample directory? All the library files need to be copied to the sample directory.

jyhflash, 2024-09-18:
> Have you copied the depthengine_2_0.dll library file to the sample directory? All the library files need to be copied to the sample directory.

This solved my problem, thanks.

closed 2024-09-26 by jyhflash