orbbec/OrbbecSDK #123: API for network discovery / listing Femto Mega devices?

ofTheo, 2024-09-16:
Hi, 

We're doing a project with 8-10 Femto Mega's. 
Is there a way to list all the Mega's connected via POE on the network? 

I know the Orbbec Viewer can do it as it shows the list in the Connect Device drop down. 
Is there a public API call to do the same thing as the Orbbec Viewer?

Thank you! 
Theo

zhonghong322, 2024-09-18:
ob::Context ctx;
    // enable net device enumration
    ctx.enableNetDeviceEnumeration(true);

    // Query the list of connected devices
    auto devList = ctx.queryDeviceList();

     // Get the number of connected devices
    if(devList->deviceCount() == 0) {
        std::cerr << "Device not found!" << std::endl;
        return -1;
    }

    for(int i = 0; i < devList->deviceCount(); i++) {
        auto sn  = devList->serialNumber(i);
        auto device = devList->getDeviceBySN(sn);

         std::shared_ptr<ob::Pipeline> pipe(new ob::Pipeline(device));
         .....
    }

zhonghong322, 2024-09-18:
You can obtain the device list in this way, and then create devices through the Serial Number (SN). You can also create devices through the device list index.

        // Create a device, 0 means the index of the first device
    auto device = devList->getDevice(0);
    std::shared_ptr<ob::Pipeline> pipe(new ob::Pipeline(device));

ofTheo, 2024-09-18:
Wow - amazing!! 

Thank you very much for the example. 
This will be really helpful.

closed 2024-09-18 by ofTheo