orbbec/OrbbecSDK #125: 加载深度引擎插件失败，初始化深度引擎失败。

xiaobai123-cell, 2024-09-18:
Hello, may I ask a question that I have been confused about for a long time. I am a Windows operating system with OpenGL version 3.1.7 and a graphics card type of NVIDA GeForce RTX 4050. My laptop has a dedicated graphics card, but when I run the depthviewer.py program on pycharm, I encounter the error 'Failed to load depth engine plugin' and 'init depth engine failed'. I have tried many methods, but it still does not work. Even after updating the latest driver, it still does not work. May I ask where the problem lies?
[image]

xiaobai123-cell, 2024-09-18:
[image]
报错界面

zhonghong322, 2024-09-18:
Have you copied the depthengine_2_0.dll library file to the sample directory?  All the library files need to be copied to the sample directory.

xiaobai123-cell, 2024-09-18:
Thank you for your answer. The problem has been resolved!

zhonghong322, 2024-09-18:
The problem has been resolved!

closed 2024-09-18 by zhonghong322