orbbec/OrbbecSDK #126: pyorbbecsdk.OBError: No required type sensor found! sensorType: OB_SENSOR_IR_LEFT

xiaobai123-cell, 2024-09-18:
[image]
Hello, when running double_infrared-viewer.py in pycharm on Windows system, the program fails to find the required sensor when attempting to obtain the flow configuration list of the left infrared sensor. How can I solve this problem?

zhonghong322, 2024-09-18:
Your Femto Bolt device doesn't support double IR, For the Python issue, please open an issue in the Python GitHub repository, https://github.com/orbbec/pyorbbecsdk  ,thanks.

xiaobai123-cell, 2024-09-18:
Thank you for your response. Are you saying that the next time you ask a question, it's not a new issue, but rather a question under someone else's issue? (Because this link is this place?)

closed 2024-09-18 by xiaobai123-cell

zhonghong322, 2024-09-19:
> Thank you for your response. Are you saying that the next time you ask a question, it's not a new issue, but rather a question under someone else's issue? (Because this link is this place?)

What I mean is that the Python GitHub repository is maintained separately from the C++ one. So, for Python issues, you should create an issue in the Python repository. This way, it can help more people.