orbbec/OrbbecSDK #128: "Depth engine create and initialize failed" after I changed AlignMode to software

bowieshi, 2024-09-18:
Hi. Here is my system information:
Computer System: Ubuntu 22.04
Device: Orbbec femto bolt
I used the following code part to align color image and depth image and show them in a window viewer, then save them to the disk.
`
    config->enableStream(depthProfile);

    config->setAlignMode(ALIGN_D2C_HW_MODE);
    pipeline.enableFrameSync();
// Start the pipeline with config
    pipeline.start(config);

    int frameCount = -10;
    Window app("viewer", colorProfile->width(), colorProfile->height(), RENDER_OVERLAY);
    while(true) {
        keyEventProcess(app, pipeline, config);
        // Wait for up to 100ms for a frameset in blocking mode.
        auto frameset = pipeline.waitForFrames(100);
        if(frameset == nullptr) {
            std::cout << "The frameset is null!" << std::endl;
            continue;
        }

        // Filter the first 5 frames of data, and save it after the data is stable
        frameCount++;
        if(frameCount < 0) {
            continue;
        }

        // Get color and depth frames
        auto colorFrame = frameset->colorFrame();
        auto depthFrame = frameset->depthFrame();
        if(colorFrame != nullptr && depthFrame != nullptr) {
            app.addToRender({ colorFrame, depthFrame });
        }
        if(colorFrame != nullptr) {
            // save the colormap
            if(colorFrame->format() != OB_FORMAT_RGB) {
                if(colorFrame->format() == OB_FORMAT_MJPG) {
                    formatConvertFilter.setFormatConvertType(FORMAT_MJPG_TO_RGB888);
                }
                else if(colorFrame->format() == OB_FORMAT_UYVY) {
                    formatConvertFilter.setFormatConvertType(FORMAT_UYVY_TO_RGB888);
                }
                else if(colorFrame->format() == OB_FORMAT_YUYV) {
                    formatConvertFilter.setFormatConvertType(FORMAT_YUYV_TO_RGB888);
                }
                else {
                    std::cout << "Color format is not support!" << std::endl;
                    continue;
                }
                colorFrame = formatConvertFilter.process(colorFrame)->as<ob::ColorFrame>();
            }
            formatConvertFilter.setFormatConvertType(FORMAT_RGB888_TO_BGR);
            colorFrame = formatConvertFilter.process(colorFrame)->as<ob::ColorFrame>();
            std::string fileName = rootPath + dirName + "/images/frame_" + formatNumber(frameCount) + ".jpg";
            saveColor(colorFrame, fileName);
            colorCount++;
        }
        else {
            std::cout << "Color frame is null!" << std::endl;
        }

        if(depthFrame != nullptr) {
            // save the depth map
            std::string fileName = rootPath + dirName + "/depth/depth_frame_" + formatNumber(frameCount) + ".png";
            saveDepth(depthFrame, fileName);
            depthCount++;
        }
        else {
            std::cout << "Depth frame is null!" << std::endl;
        }
`
I found the alignment is not very accurate and have latency in the viewer. So I tried to modify the align model from hardware to software.
my modification is following:
`
    config->setAlignMode(ALIGN_D2C_SW_MODE);
`
Then I compiled and ran the code again. However I can not read the depth image data stream. The viewer is in black. I use the official SDK Orbbec Viewer. However I can not see the depth image as well and I got the following error:
`[09/18 23:09:15.464785][info][79870][VideoSensor.cpp:646] start OB_SENSOR_DEPTH stream with profile: {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 15}, Backend Format: OB_FORMAT_Y16
[09/18 23:09:15.464809][info][79870][MSDEConverterDevice.cpp:549] Start real profile,width:7680 height:434
[09/18 23:09:15.468892][info][79989][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:490904
[09/18 23:09:15.468940][info][79989][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
GlContext::Initialize eglInitialize error code = 0x3001
[09/18 23:09:15.555822][error][79989][MSDEConverterDevice.cpp:101] Depth engine create and initialize failed,retCode:204
[09/18 23:09:15.555896][info][79989][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:490904
[09/18 23:09:15.555900][info][79989][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
GlContext::Initialize eglInitialize error code = 0x3001
[09/18 23:09:15.629072][error][79989][MSDEConverterDevice.cpp:101] Depth engine create and initialize failed,retCode:204
[09/18 23:09:15.629144][info][79989][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:490904
[09/18 23:09:15.629148][info][79989][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
`
Even I change the align model back to harware and run again. The problem is still here. I have no idea how to get the depth data stream back. I wonder that is some setting inside the camera  modified? If you can help, I would be more than grateful.
[image]

zhonghong322, 2024-09-19:
Based on the logs, the depth engine initialization failure is unrelated to the D2C settings. You can use the Orbbec Viewer to open the depth stream separately for cross-verification. 

This issue is related to the OpenGL version. Is your computer equipped with a dedicated graphics card? Please provide a screenshot of the graphics card driver or try upgrading the graphics card driver.

The following log is from within the depth engine library.

libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen

bowieshi, 2024-09-19:
> Based on the logs, the depth engine initialization failure is unrelated to the D2C settings. You can use the Orbbec Viewer to open the depth stream separately for cross-verification.
>
> This issue is [...]

Thank you for your reply. Today I reboot the linux and replug in the camera. The depth channel return to normal.

closed 2024-09-19 by bowieshi