orbbec/OrbbecSDK #132: Disable log file

Allius27, 2024-09-20:
Camera - Gemini 2
SDK version - v1.10.12
Operation system - Ubuntu20

Hi there

How to disable saving log "Log/OrbbecSDK.log.txt" ?

zhonghong322, 2024-09-21:
1、You can modify the configuration file to disable logging.
[image]
2、If you want to implement it with code, please refer to the documentation.
https://orbbec.github.io/OrbbecSDK/doc/tutorial/English/OrbbecSDK_C++_API_user_guide-v1.0.pdf

zhonghong322, 2024-09-27:
@Allius27 Have your questions been resolved?

Allius27, 2024-09-27:
Yes, thank. 

For others 
Completly log might be disable via "OrbbecSDKConfig_v1.0.xml"
And file even not creates.

Disabling it from the API doesn't work (without config file)
Log fIle creates anyway, but doesn't fill after API disabling
```
ob::Context m_ctx;
m_ctx.setLoggerToFile(OB_LOG_SEVERITY_OFF, NULL);
```

closed 2024-09-27 by Allius27