orbbec/OrbbecSDK #133: SDK unable to see gemini E

Allius27, 2024-09-20:
Camera - Gemini E
SDK version - v1.10.12
Operation system - Ubuntu20 / Windows10

Hi there

I got Gemini E example and SDK unable to connect this camera under ubuntu20 ( under wmware ) and windows10 ( native OS )

ubuntu20 ( under wmware )
[image]

windows10 ( native OS )
[image]

And i noticed that gemini E represented in wmware as 2 devices, howerer gemini 2 represented as single

Gemini E
[image]

Gemini 2 
[image]

What's wrong with a Gemini E ?

zhonghong322, 2024-09-21:
[image]
[image]
I tested it, and it works fine. Please use lsusb to check if the device's PID is enumerated.

Allius27, 2024-09-23:
```wmware ubuntu20 dmesg -w```
[image]

```wmware ubuntu20 lsusb```
[image]

```hardware ubuntu20 dmesg -w```
[image]

```hardware ubuntu20 lsusb```
[image]

```hardware ubuntu20 ./OBHelloOrbbec```
[image]

For the current moment looks, that "Gemini E" firmware incompatible with "VMware Workstation 17 Player" version.
Which version do you use?

zhonghong322, 2024-09-23:
> Referenc
1.16.2

closed 2024-09-27 by Allius27