orbbec/OrbbecSDK #135: Femto bolt 多设备同步录制问题

jyhflash, 2024-09-26:
实现两台Femto bolt同步录制。
首先我分别从k4aviewer上将从机和主机分别运行，可以看到两台设备是同步的。
然后进行录制工作，我使用k4arecorder.exe运行命令行进行录制。
首先对于从机设备打开命令行运行k4arecorder.exe --device 0 --external-sync subordinate --imu OFF -e 16000 -r 15 -l 10 output-2.mkv
会显示[subordinate mode] Waiting for signal from master并等待主设备录制开启。
然后对于主机设备运行k4arecorder.exe --device 1 --external-sync master --imu OFF -e 16000 -r 15 -l 10 output-1.mkv
此时主设备还在初始化还没开始录制，而从机设备已经开始录制了，显示Started recording，主机此时还在初始化，过了两三秒，主机初始化完才显示Started recording，导致两台设备录制不同步，从机设备总是提前录制完。
随后查看录制保存的视频显示的总是从机提前主机开始录制。并且，一机一台电脑和两机一台电脑两种连接都会出现这个问题。
这好像是从机只要接收到主机启动的脉冲就开始录制，但是此时主机还没完全启动完，也就没办法录制。从而导致两台设备没办法同时录制。
请问有什么解决方法吗？

zhonghong322, 2024-09-27:
@jyhflash Please submit issues related to the K4A wrapper in the https://github.com/orbbec/OrbbecSDK-K4A-Wrapper repository. We have dedicated personnel managing each repository. Thank you!

zhonghong322, 2024-09-29:
已经在K4a wrapper https://github.com/orbbec/OrbbecSDK-K4A-Wrapper/issues/91 提issue，该issue关闭

closed 2024-09-29 by zhonghong322