orbbec/OrbbecSDK #140: Depth camera stops working

hupba-scanner, 2024-10-11:
I'm working on a setup where I run multiple (up to 8) Bolts synchronized and triggered by an external trigger. I have the setup working, but after some time, some of the Bolts started developing problems. They can only use the color camera and have no output on the depth/ir cameras. I tried it from the SDK (v1.10.12) and OrbbecViewer as well. When running the code I get the following error in the logs (this is the output of the DepthViewer example). What does this error code mean? Is it a software or hardware issue?

Some more context: other Bolts work well when using them alone or with other Bolts, the problematic ones don't work in either case, even when they are the only device plugged in.

`
[10/11 13:34:18.343238][info][28785][Context.cpp:68] Context created with config: /home/scanner-admin/workspace/OrbbecSDK_v1.10.12/Example/build/bin/OrbbecSDKConfig_v1.0.xml
[10/11 13:34:18.343251][info][28785][Context.cpp:73] Work directory=/home/scanner-admin/workspace/OrbbecSDK_v1.10.12/Example/build/bin, SDK version=v1.10.12-20240809-69b687f
[10/11 13:34:18.343260][info][28785][LinuxPal.cpp:32] createObPal: create LinuxPal!
[10/11 13:34:18.577432][info][28785][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[10/11 13:34:18.577460][info][28785][DeviceManager.cpp:15] Current found device(s): (1)
[10/11 13:34:18.577466][info][28785][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL865410368, Connection: USB3.1
[10/11 13:34:18.577476][info][28785][Pipeline.cpp:15] Try to create pipeline with default device.
[10/11 13:34:18.577516][info][28785][FemtoBoltUvcDevice.cpp:23] FemtoBoltUvcDevice init ...
[10/11 13:34:18.577573][info][28785][FemtoBoltUvcDevice.cpp:122] Create command start!
[10/11 13:34:18.579040][info][28785][MSDEConverterDevice.cpp:726] Succeed to load depth engine plugin
[10/11 13:34:18.785902][info][28785][AbstractDevice.cpp:121] 	- Firmware version: 1.0.9
[10/11 13:34:18.785933][info][28785][FemtoBoltUvcDevice.cpp:281] Create command done!
[10/11 13:34:18.785946][info][28785][FemtoBoltUvcDevice.cpp:402] init sensor map start!
[10/11 13:34:18.786006][info][28785][FemtoBoltUvcDevice.cpp:429] init sensor map done!
[10/11 13:34:18.786143][info][28785][FemtoBoltUvcDevice.cpp:285] Init depth process param start!
[10/11 13:34:19.131295][info][28785][MSDEConverterDevice.cpp:777] got nvram data succeed.
[10/11 13:34:19.446811][info][28785][FemtoBoltUvcDevice.cpp:359] setNvramDataStreamStopFunc succeed
[10/11 13:34:19.446899][info][28785][FemtoBoltUvcDevice.cpp:398] Init depth process param done!
[10/11 13:34:19.447902][info][28785][FemtoBoltUvcDevice.cpp:39] FemtoBoltUvcDevice init done!
[10/11 13:34:19.447928][info][28785][DeviceManager.cpp:150] Device created successfully! Name: Femto Bolt, PID: 0x066b, SN/ID: CL865410368
[10/11 13:34:19.447967][info][28785][Pipeline.cpp:47] Pipeline created with device: {name: Femto Bolt, sn: CL865410368}, @0x57D932466C20
[10/11 13:34:19.448007][info][28785][Pipeline.cpp:188] Check and set config start!
[10/11 13:34:19.448822][info][28785][FemtoBoltUvcDevice.cpp:529] Depth sensor has been created!
[10/11 13:34:19.448885][info][28785][VideoSensor.cpp:384] Query stream profile! size= 14, SensorType=OB_SENSOR_DEPTH
[10/11 13:34:19.448900][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 15}
[10/11 13:34:19.448911][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 1024, height: 1024, fps: 15}
[10/11 13:34:19.448921][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 1024, height: 1024, fps: 5}
[10/11 13:34:19.448932][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 30}
[10/11 13:34:19.448942][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 30}
[10/11 13:34:19.448952][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 25}
[10/11 13:34:19.448962][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 25}
[10/11 13:34:19.448972][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 15}
[10/11 13:34:19.448982][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 5}
[10/11 13:34:19.448993][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 5}
[10/11 13:34:19.449004][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 30}
[10/11 13:34:19.449015][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 25}
[10/11 13:34:19.449043][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 15}
[10/11 13:34:19.449053][info][28785][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 5}
[10/11 13:34:19.449069][info][28785][Pipeline.cpp:223] Check and set config done!
[10/11 13:34:19.449638][info][28785][Pipeline.cpp:288] Try to start streams!
[10/11 13:34:19.449729][info][28785][VideoSensor.cpp:646] start OB_SENSOR_DEPTH stream with profile: {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 15}, Backend Format: OB_FORMAT_Y16
[10/11 13:34:19.449742][info][28785][MSDEConverterDevice.cpp:549] Start real profile,width:7680 height:434
[10/11 13:34:19.452179][info][28811][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:1048576
[10/11 13:34:19.452213][info][28811][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
[10/11 13:34:19.452527][error][28811][MSDEConverterDevice.cpp:101] Depth engine create and initialize failed,retCode:103
[10/11 13:34:19.452623][info][28811][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:1048576
[10/11 13:34:19.452635][info][28811][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
[10/11 13:34:19.452735][error][28811][MSDEConverterDevice.cpp:101] Depth engine create and initialize failed,retCode:103
[10/11 13:34:19.452819][info][28811][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:1048576
[10/11 13:34:19.452831][info][28811][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
[10/11 13:34:19.452970][error][28811][MSDEConverterDevice.cpp:101] Depth engine create and initialize failed,retCode:103
[10/11 13:34:19.453028][info][28811][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:1048576
[10/11 13:34:19.453035][info][28811][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
[10/11 13:34:19.453100][error][28811][MSDEConverterDevice.cpp:101] Depth engine create and initialize failed,retCode:103
[10/11 13:34:19.453163][info][28811][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:1048576
[10/11 13:34:19.453169][info][28811][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
[10/11 13:34:19.453223][error][28811][MSDEConverterDevice.cpp:101] Depth engine create and initialize failed,retCode:103
[10/11 13:34:19.453288][info][28811][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:1048576
[10/11 13:34:19.453294][info][28811][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
[10/11 13:34:19.453346][error][28811][MSDEConverterDevice.cpp:101] Depth engine create and initialize failed,retCode:103
[10/11 13:34:19.453410][info][28811][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:1048576
[10/11 13:34:19.453416][info][28811][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
[10/11 13:34:19.453469][error][28811][MSDEConverterDevice.cpp:101] Depth engine create and initialize failed,retCode:103
[10/11 13:34:19.453529][info][28811][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:1048576
[10/11 13:34:19.453534][info][28811][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
[... 4 lines cut ...]
[10/11 13:34:19.453772][info][28811][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:1048576
[10/11 13:34:19.453776][info][28811][MSDEConverterDevice.cpp:94] use dynlib load depthengine lib......
[... cut ...]
[10/11 13:34:34.133092][info][28785][VideoSensor.cpp:303] VideoSensor destroyed, @OB_SENSOR_DEPTH
[10/11 13:34:34.134044][info][28785][ObLibuvcDevicePort.cpp:70] uvc_close done.
[10/11 13:34:34.134074][info][28785][ObLibuvcDevicePort.cpp:71] ~ObLibuvcDevicePort done
[10/11 13:34:34.134453][info][28785][Context.cpp:84] Context destroyed
[10/11 13:34:34.134546][warning][29075][Pipeline.cpp:377] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**53 logs in 5573ms, last: 13:34:33.865390**]

`

zhonghong322, 2024-10-14:
How many cameras have this issue? After the problem occurs, if you power off the camera and then power it back on, does the issue still persist?

zhonghong322, 2024-10-14:
This error is caused by an exception when reading parameters from the camera.

hupba-scanner, 2024-10-14:
3 out of 8 cameras developed this problem. The problem persists even after restarting. I tried it with different cables as well. Is it possible that it's a hardware issue?

Somewhere I remember seeing that it's not recommended to write the MultiDeviceSyncConfig to the device too often so that the flash memory doesn't stop working. And I have done this only a very few times. But could this be a possible cause?

shivanshpatel35, 2025-04-02:
Same issue