orbbec/OrbbecSDK #141: Can not read depth stream when using USB Extension cable

bowieshi, 2024-10-14:
Hi. I am using a USB Extension Cable to link my Orbbec femto bolt depth camera to my ubuntu computer. 
When I run the code under the `Sample_DepthViewer` directory, I can read the depth information, like the following:
[image]
I think the USB Extension cable is OK for transmitting the depth information.
However, when I run the code under the `Sample_SaveToDisk`, I can not read depth stream and can not save the depth images successfully, like the following:
[image]
When I do not use USB Extension cable and directly use the officially provided cable to link my camera to computer, I can use code under the `Sample_SaveToDisk` to read in depth stream, like following:
[image]
The color stream is OK anytime, both `Sample_ColorViewer` and `Sample_SaveToDisk`, the color stream is normal and I can read it. 
I can not find what is the potential problem of the depth stream. Could you help me out? Thank you. I am using this cable to test: 
[image]

zhonghong322, 2024-10-17:
https://www.amazon.com/Compatible-Kuject-Transfer-Charging-%20Chargers/dp/B09DSD8RZL/?th=1    This is the USB extension cable that we have tested.

bowieshi, 2024-10-17:
@zhonghong322 thank you for your response. Can I inquire whether you have tested any USB extension purchasing available within mainland? It seems that buying Kuject Link USB extension is not very convenient in mainland. Thank you very much!