orbbec/OrbbecSDK #142: Cannot complete registering metadata in Powershell

triplecmike, 2024-10-17:
Hi,

I'm very new to Femto Mega and programming stuff.
I'm trying to register metadata following this post: https://github.com/orbbec/OrbbecSDK/blob/main/misc/scripts/obsensor_metadata_win10.md 
It failed and returned the following error message in Powershell:

```
.\obsensor_metadata_win10.ps1 : File C:\OrbbecSDK_v1.10.12\script\obsensor_metadata_win10.ps1 cannot be loaded. The
file C:\OrbbecSDK_v1.10.12\script\obsensor_metadata_win10.ps1 is not digitally signed. You cannot run this script on
the current system. For more information about running scripts and setting execution policy, see
about_Execution_Policies at https:/go.microsoft.com/fwlink/?LinkID=135170.
At line:1 char:1
+ .\obsensor_metadata_win10.ps1 -op install_all
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : SecurityError: (:) [], PSSecurityException
    + FullyQualifiedErrorId : UnauthorizedAccess
```

Could anyone shed some light on this for me? Thanks!

zhonghong322, 2024-10-18:
Do you run PowerShell with administrator privileges? You can also run Orbbec Viewer, as this tool will automatically perform the registration of metadata.

triplecmike, 2024-10-21:
"@zhonghong322 Thanks for getting back to me so quickly. Yes, I did run PowerShell with 'Run as administrator.' I've also run Orbbec Viewer a couple of times already. Does this mean the metadata registration is complete after running Orbbec Viewer, and that I no longer need to run the 'script obsensor_metadata_win10.ps1'? Thanks!

closed 2024-11-19 by triplecmike