orbbec/OrbbecSDK #144: Compile "Sample-PointCloud" in Visual Studio

triplecmike, 2024-10-23:
Hi,

I'm new to this device and C++ and have a quite basic question. I was trying to compile and run the sample program ```Sample-PointCloud``` in Visual Studio.
I've added directories through Project -> Properties:
VC++ Directories
```
C:\OrbbecSDK_v1.10.12\SDK\include
C:\OrbbecSDK_v1.10.12\SDK\lib
```
Linker -> Input
```
OrbbecSDK.lib
```
[image]
Many errors popped out when I tried debugging and built the project:
```
Severity	Code	Description	Project	File	Line	Suppression State	Details
Error (active)	E0020	identifier "kbhit" is undefined	PointCloudTest	C:\Users\mchang\source\repos\PointCloudTest\PointCloudTest\Source.cpp	152		
Error (active)	E0020	identifier "getch" is undefined	PointCloudTest	C:\Users\mchang\source\repos\PointCloudTest\PointCloudTest\Source.cpp	153		
Warning	C4251	'ob::Context::impl_': 'std::unique_ptr<ContextImpl,std::default_delete<ContextImpl>>' needs to have dll-interface to be used by clients of 'ob::Context'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Context.hpp	23		
Warning	C4251	'std::enable_shared_from_this<ob::Filter>::_Wptr': 'std::weak_ptr<_Ty>' needs to have dll-interface to be used by clients of 'std::enable_shared_from_this<ob::Filter>'
        with
        [
            _Ty=ob::Filter
        ]	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Filter.hpp	27		
Warning	C4251	'ob::Filter::impl_': 'std::shared_ptr<FilterImpl>' needs to have dll-interface to be used by clients of 'ob::Filter'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Filter.hpp	103		
Warning	C4251	'ob::Filter::type_': 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of 'ob::Filter'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Filter.hpp	104		
Warning	C4251	'ob::Device::impl_': 'std::unique_ptr<DeviceImpl,std::default_delete<DeviceImpl>>' needs to have dll-interface to be used by clients of 'ob::Device'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Device.hpp	31		
Warning	C4251	'ob::DeviceInfo::impl_': 'std::unique_ptr<DeviceInfoImpl,std::default_delete<DeviceInfoImpl>>' needs to have dll-interface to be used by clients of 'ob::DeviceInfo'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Device.hpp	642		
Warning	C4251	'ob::DeviceList::impl_': 'std::unique_ptr<DeviceListImpl,std::default_delete<DeviceListImpl>>' needs to have dll-interface to be used by clients of 'ob::DeviceList'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Device.hpp	759		
Warning	C4251	'ob::CameraParamList::impl_': 'std::unique_ptr<CameraParamListImpl,std::default_delete<CameraParamListImpl>>' needs to have dll-interface to be used by clients of 'ob::CameraParamList'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Device.hpp	869		
Warning	C4251	'ob::OBDepthWorkModeList::impl_': 'std::unique_ptr<OBDepthWorkModeListImpl,std::default_delete<OBDepthWorkModeListImpl>>' needs to have dll-interface to be used by clients of 'ob::OBDepthWorkModeList'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Device.hpp	896		
Warning	C4251	'ob::DevicePresetList::impl_': 'std::unique_ptr<DevicePresetListImpl,std::default_delete<DevicePresetListImpl>>' needs to have dll-interface to be used by clients of 'ob::DevicePresetList'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Device.hpp	940		
Warning	C4251	'ob::Error::impl_': 'std::unique_ptr<ErrorImpl,std::default_delete<ErrorImpl>>' needs to have dll-interface to be used by clients of 'ob::Error'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Error.hpp	16		
Warning	C4251	'std::enable_shared_from_this<ob::Frame>::_Wptr': 'std::weak_ptr<_Ty>' needs to have dll-interface to be used by clients of 'std::enable_shared_from_this<ob::Frame>'
        with
        [
            _Ty=ob::Frame
        ]	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Frame.hpp	42		
Warning	C4251	'ob::Frame::impl_': 'std::unique_ptr<FrameImpl,std::default_delete<FrameImpl>>' needs to have dll-interface to be used by clients of 'ob::Frame'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Frame.hpp	44		
Warning	C4251	'ob::Pipeline::impl_': 'std::unique_ptr<PipelineImpl,std::default_delete<PipelineImpl>>' needs to have dll-interface to be used by clients of 'ob::Pipeline'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Pipeline.hpp	30		
Warning	C4251	'ob::Config::impl_': 'std::unique_ptr<ConfigImpl,std::default_delete<ConfigImpl>>' needs to have dll-interface to be used by clients of 'ob::Config'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Pipeline.hpp	212		
Warning	C4251	'ob::Recorder::impl_': 'std::unique_ptr<RecorderImpl,std::default_delete<RecorderImpl>>' needs to have dll-interface to be used by clients of 'ob::Recorder'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\RecordPlayback.hpp	25		
Warning	C4251	'ob::Playback::impl_': 'std::unique_ptr<PlaybackImpl,std::default_delete<PlaybackImpl>>' needs to have dll-interface to be used by clients of 'ob::Playback'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\RecordPlayback.hpp	65		
Warning	C4251	'ob::Sensor::impl_': 'std::unique_ptr<SensorImpl,std::default_delete<SensorImpl>>' needs to have dll-interface to be used by clients of 'ob::Sensor'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Sensor.hpp	33		
Warning	C4251	'ob::SensorList::impl_': 'std::unique_ptr<SensorListImpl,std::default_delete<SensorListImpl>>' needs to have dll-interface to be used by clients of 'ob::SensorList'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Sensor.hpp	82		
Warning	C4251	'ob::OBFilterList::impl_': 'std::unique_ptr<OBFilterListImpl,std::default_delete<OBFilterListImpl>>' needs to have dll-interface to be used by clients of 'ob::OBFilterList'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\Sensor.hpp	127		
Warning	C4251	'std::enable_shared_from_this<ob::StreamProfile>::_Wptr': 'std::weak_ptr<_Ty>' needs to have dll-interface to be used by clients of 'std::enable_shared_from_this<ob::StreamProfile>'
        with
        [
            _Ty=ob::StreamProfile
        ]	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\StreamProfile.hpp	22		
Warning	C4251	'ob::StreamProfile::impl_': 'std::unique_ptr<StreamProfileImpl,std::default_delete<StreamProfileImpl>>' needs to have dll-interface to be used by clients of 'ob::StreamProfile'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\StreamProfile.hpp	24		
Warning	C4251	'ob::StreamProfileList::impl_': 'std::unique_ptr<StreamProfileListImpl,std::default_delete<StreamProfileListImpl>>' needs to have dll-interface to be used by clients of 'ob::StreamProfileList'	PointCloudTest	C:\OrbbecSDK_v1.10.12\SDK\include\libobsensor\hpp\StreamProfile.hpp	214		
Error	C4996	'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	PointCloudTest	C:\Users\mchang\source\repos\PointCloudTest\PointCloudTest\Source.cpp	15		
Error	C4996	'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	PointCloudTest	C:\Users\mchang\source\repos\PointCloudTest\PointCloudTest\Source.cpp	37		
Error	C3861	'kbhit': identifier not found	PointCloudTest	C:\Users\mchang\source\repos\PointCloudTest\PointCloudTest\Source.cpp	152		
Error	C3861	'getch': identifier not found	PointCloudTest	C:\Users\mchang\source\repos\PointCloudTest\PointCloudTest\Source.cpp	153		
```
And here's the original code in the sample folder:
```
#include "libobsensor/ObSensor.hpp"
// #include "opencv2/opencv.hpp"
#include <fstream>
#include <iostream>
// #include "utils.hpp"
#include "libobsensor/hpp/Utils.hpp"

#define KEY_ESC 27
#define KEY_R 82
[... 192 lines cut ...]
                            std::shared_ptr<ob::Frame> frame = pointCloud.process(frameset);
                            savePointsToPly(frame, "DepthPoints.ply");
                            std::cout << "DepthPoints.ply Saved" << std::endl;
                        }
                        catch (std::exception& e) {
                            std::cout << "Get point cloud failed" << std::endl;
                        }
                        break;
                    }
                }
            }
        }
    }
    // stop the pipeline
    pipeline.stop();

    return 0;
}
catch (ob::Error& e) {
    std::cerr << "function:" << e.getName() << "\nargs:" << e.getArgs() << "\nmessage:" << e.getMessage() << "\ntype:" << e.getExceptionType() << std::endl;
    exit(EXIT_FAILURE);
}
```

Could anyone help me with this? Thank you very much!

zhonghong322, 2024-10-24:
You can refer to this document to compile the sample. https://github.com/orbbec/OrbbecSDK/blob/main/doc/tutorial/English/Environment_Configuration.md

triplecmike, 2024-10-24:
Thank you so much. this is very helpful.

closed 2024-11-19 by triplecmike