orbbec/OrbbecSDK #145: Orbbec Viewer not detecting depth camera and not having any functions on the screen

noellim040318, 2024-10-24:
[image]

zhonghong322, 2024-10-24:
OrbbecSDK now has two versions, v1.x.x and v2.x.x. You are currently using version v2.x, which is the open-source version. 
Its GitHub repository is: https://github.com/orbbec/OrbbecSDK-dev. For future versions of v2.x, please submit issues in this repository. 
What device are you using? v2.x only supports the following devices， Please check if your device is on the supported list.

[image]

noellim040318, 2024-10-24:
Oh so it doesn't support orbbec astra pro plus? But even v1.x.x also its
not working. Nothing is showing up on the screen, not even the device

valan01, 2025-02-11:
did u find any solution. i have same issue. i can see the outrput on openni viewer. but orbbec viewer still not detecting anything

saneng-nijoichi, 2025-05-25:
I have same issue. I can see the output on openni viewer as well, but orbbec viewer still not detecting anything
I tried all versions of the viewer available for

zhonghong322, 2025-05-29:
> Oh so it doesn't support orbbec astra pro plus? But even v1.x.x also its
> not working. Nothing is showing up on the screen, not even the device
> […](#)

Yes，some old openni device not support.